kirillzyusko/react-native-wifi-p2p #1: Example in documentation is incorrect

joshmbg, 2018-08-29:
It refers to react-native-android-p2p-wifi, when the package is actually named react-native-wifi-p2p.

When I try running the example, it doesn't work, and I don't know if this is because the example is incorrect or if there is something wrong with my test setup - the example needs to be reviewed and updated to ensure it is correct.

kirillzyusko, 2018-08-29:
@joshmbg Thanks for your remark. I really forget to rename this package in readme (already fixed it).
But these code examples are working, and I tested it in my own project. Could you provide more specific info about your environment, like device (emulator), which method is not working and so on?
BTW, I have test repository, where I created button for each method. And I tested it on real devices and all works perfectly. If you needed this project, give me know and I'll push it here.

joshmbg, 2018-08-30:
Don't worry about it, I tested it on my real phone and it worked, so something's wrong with my emulator setup not with the code example.

closed 2018-08-30 by joshmbg

kirillzyusko, 2018-08-30:
@joshmbg On any emulators you don't have access to native modules such as wi-fi/bluetooth/camera and so on. So, libraries, that provide access to native modules, should test on real devices.
In any case, thanks for your feedback. I think, that I should write info about testing in readme :)