kirillzyusko/react-native-wifi-p2p #2: Execution failed for task ':react-native-wifi-p2p:verifyReleaseResources'.

lokixviii, 2018-10-26:
Hi, everytime I try to build my app it always fails:

> Configure project :react-native-wifi-p2p
WARNING: Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'.
It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html
WARNING: The specified Android SDK Build Tools version (23.0.1) is ignored, as it is below the minimum supported version (27.0.3) for Android Gradle Plugin 3.1.4.
Android SDK Build Tools 27.0.3 will be used.
To suppress this warning, remove "buildToolsVersion '23.0.1'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.

> Task :app:bundleReleaseJsAndAssets
warning: the transform cache was reset.
Loading dependency graph, done.
bundle: Writing bundle output to: C:\Users\darrell\dev\idverifier\android\app\build\generated\assets\react\release\index.android.bundle
bundle: Done writing bundle output

error: resource android:style/TextAppearance.Material.Widget.Button.Borderless.Colored not found.
error: resource android:style/TextAppearance.Material.Widget.Button.Colored not found.
C:\Users\darrell\.gradle\caches\transforms-1\files-1.1\appcompat-v7-27.1.1.aar\4128735aae96be8c7749e766b675672b\res\values-v26\values-v26.xml:9:5-12:13: AAPT: error: resource android:attr/colorError not found.

C:\Users\darrell\.gradle\caches\transforms-1\files-1.1\appcompat-v7-27.1.1.aar\4128735aae96be8c7749e766b675672b\res\values-v26\values-v26.xml:13:5-16:13: AAPT: error: resource android:attr/colorError not found.

C:\Users\darrell\.gradle\caches\transforms-1\files-1.1\appcompat-v7-27.1.1.aar\4128735aae96be8c7749e766b675672b\res\values-v26\values-v26.xml:17:5-93: AAPT: error: style attribute 'android:attr/keyboardNavigationCluster' not found.

C:\Users\darrell\.gradle\caches\transforms-1\files-1.1\appcompat-v7-27.1.1.aar\4128735aae96be8c7749e766b675672b\res\values\values.xml:251:5-69: AAPT: error: resource android:attr/fontStyle not found.

C:\Users\darrell\.gradle\caches\transforms-1\files-1.1\appcompat-v7-27.1.1.aar\4128735aae96be8c7749e766b675672b\res\values\values.xml:251:5-69: AAPT: error: resource android:attr/font not found.

C:\Users\darrell\.gradle\caches\transforms-1\files-1.1\appcompat-v7-27.1.1.aar\4128735aae96be8c7749e766b675672b\res\values\values.xml:251:5-69: AAPT: error: resource android:attr/fontWeight not found.

error: failed linking references.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-wifi-p2p:verifyReleaseResources'.
> com.android.ide.common.process.ProcessException: Failed to execute aapt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3m 0s
70 actionable tasks: 57 executed, 13 up-to-date

-------------------------------------------------------------------------------------------------
I'm using
"react": "16.6.0-alpha.8af6728",
"react-native": "0.57.4",
"react-native-wifi-p2p": "^0.6.3"

Please help me fix this. Thanks!

kirillzyusko, 2018-10-26:
@lokixviii Could you try to add next lines to android/build.gradle and try to compile project again?

`
allprojects { ... }

subprojects {
    project.configurations.all {
        afterEvaluate {project ->
            if (project.hasProperty("android")) {
                android {
                    compileSdkVersion 26
                    buildToolsVersion '26.0.2'
                }
            }
        }
        resolutionStrategy.eachDependency { details ->
            if (details.requested.group == 'com.android.support'
                    && !details.requested.name.contains('multidex') ) {
                details.useVersion "26.0.2"
            }
        }
    }
}
`

Avantica777, 2019-04-08:
Thanks for posting about this issue.
@kirillzyusko, I've the same issue and I tried like above, but it's not working on my side.

`
allprojects {
    repositories {
        mavenLocal()
        google()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
    }
}

subprojects {
    project.configurations.all {
        afterEvaluate {project ->
            if (project.hasProperty("android")) {
                android {
                    compileSdkVersion 27
                    buildToolsVersion '28.0.3'
                }
            }
        }
        resolutionStrategy.eachDependency { details ->
            if (details.requested.group == 'com.android.support'
                    && !details.requested.name.contains('multidex') ) {
                details.useVersion "28.0.3"
            }
        }
    }
}`

I tried like above and I get:
_$ ./gradlew assembleRelease

> Configure project :app
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)
registerResGeneratingTask is deprecated, use registerGeneratedResFolders(FileCollection)

FAILURE: Build failed with an exception.

* Where:
Build file 'D:\Work\hybrid\FlapSentry_Design3.13\android\build.gradle' line: 38

* What went wrong:
A problem occurred configuring project ':app'.
> Could not create task ':app:mergeDebugResources'.
   > Project#afterEvaluate(Closure) on project ':app' cannot be executed in the current context.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/4.10.1/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 1s_

and 38 line cod is
`afterEvaluate {project ->`

kirillzyusko, 2019-04-09:
@Avantica777 you can look at working example project: https://github.com/kirillzyusko/react-native-wifi-p2p-example
I suppose you need to synchronize gradle version between your project and my. It's common error that you may face with any library. So I'd recommend you just search this problem, and I suppose you find cause and will be able to fix it.