kirillzyusko/react-native-wifi-p2p #3: How do peers communicate with each other?

bharadwajpro, 2018-10-27:
Connecting, Disconnecting, forming a group everything is fine. But how do peers exchange information? `sendMessage()` api is not available as of now according to `README`

kirillzyusko, 2018-11-03:
@bharadwajpro Could you test last changes from `master` branch?
Just add the following line to your `package.json` and run `npm install`:

```"react-native-wifi-p2p": "github:kirikzyusko1996/react-native-wifi-p2p#master"```

bharadwajpro, 2018-11-14:
I was neither able to form a group nor connect to a device. It says `Error code 2: Operation failed because the framework is busy and unable to service the request.`. So I wasn't able to test your changes. May be there is a problem with devices I am using.

kirillzyusko, 2018-11-18:
@bharadwajpro I'll close this issue, since I tested message sending/receiving and it works fine on my devices. If you will provide more information of your problem (devices, your steps for reproducing this problem, were connected these devices to each other via `Wi-Fi direct` earlier and so on, I'll try to help you. Because `BUSY_ERROR` - it's not error of my library, it's "native" error (more info you can find [here](https://developer.android.com/reference/android/net/wifi/p2p/WifiP2pManager.html?hl=pt-BR#BUSY)). I just made a bridge between "native" code and js side.

closed 2018-11-18 by kirillzyusko