kirillzyusko/react-native-wifi-p2p #4: getAvailablePeers always returns empty array

io-pan, 2018-11-30:
1st: thanks for that awesome repo.

I tested it on my galaxy S6 and also on S9.
I'am running the same code on both phone.
Everything works fine on S6, it detects my wifi-printer, but on S9 getAvailablePeers always returns an empty array. 
Any idea  why ?

kirillzyusko, 2018-11-30:
@io-pan First of all I would like to recommend trying to detect `S6` from `S9` and `S9` from `S6` (without wifi-printer). These devices support WiFi direct mode, so you can try to enable this mode on one device, and try to find from another. And then switch role. Also I don't recommend using `getAvailablePeers`, since I added this method as temporary. Best approach for detecting devices is using subscription methods, like `subscribeOnPeersUpdates`. Callback, that passes into this methos will always call, when info about available peers has changed. I know, that it works with delay, but it's not my blame. I've only written bridge between native code and JS.
And in fact, yes, I've also faced with this problem. I have Xiaomi Redmi Note 5 Pro (android 8.1) and Prestigio 4500T (android 4.2). And `4500T` can see `Note 5 Pro`, but on `Note 5 Pro` I can't see my `4500T`. And I don't know a reason of such behaviour.

closed 2018-12-16 by kirillzyusko

rolandjitsu, 2019-06-23:
@kirillzyusko I am having the same problem ... Subscriptions also give me an empty list of devices. Running on an Google Pixel 3, latest Android.

kirillzyusko, 2019-06-23:
@rolandjitsu What is the second device, which you are trying to find? Are you sure, that WiFi direct mode is enabled on both devices?

rolandjitsu, 2019-06-23:
The second device is a Raspberry Pi 3 configured using [wpa_supplicant](https://raspberrypi.stackexchange.com/a/84535). Using Android's Wifi Direct search (it's in the phone's settings) I can see the device.

This libs is discovering but it doesn't show me the devices.

kirillzyusko, 2019-06-23:
@rolandjitsu I got it. You know I have faced with the same problem. When I have been testing my library, I've found, that I could detect my new phone from my old, but not vice versa. And from Android settings I was able to see, and there everything worked as expected. To be honest I don't know why it happened. And also I don't know even reason. Maybe I did a mistake in my code. But I don't so think, since I just copied official documentation and pasted it. And during my testing I thought it is only my specific problem. But you are not the first user, who reported me similar issue. Really I don't know even a cause of such behaviour. And I think, unfortunately, I can't help here :(

rolandjitsu, 2019-06-23:
No worries. I might not be able to use WiFi Direct for what I'm trying to do anyway. But thanks for trying to help.