kirillzyusko/react-native-wifi-p2p #6: App crashes on starting up

ekxz900, 2019-02-04:
Hi,

I'm getting this error on successful build. Can you please help?

Attempt to invoke virtual method 'android.net.wifi.p2p.WifiP2pManager$Channel android.net.wifi.p2p.WifiP2pManager.initialize(android.content.Context, android.os.Looper, android.net.wifi.p2p.WifiP2pManager$ChannelListener)' on a null object reference
init
    WiFiP2PManagerModule.java:81
invoke
    Method.java
invoke
    JavaMethodWrapper.java:372
invoke
    JavaModuleWrapper.java:160
run
    NativeRunnable.java
handleCallback
    Handler.java:790
dispatchMessage
    Handler.java:99
dispatchMessage
    MessageQueueThreadHandler.java:29
loop
    Looper.java:164
run
    MessageQueueThreadImpl.java:192
run
    Thread.java:764

kirillzyusko, 2019-02-04:
@ekxz900 Hi, are you sure that you use **real** device (not emulator) with p2p support?

ekxz900, 2019-02-04:
Hi @kirillzyusko, ahhh my mistake so sorry..

closed 2019-02-04 by ekxz900