kirillzyusko/react-native-wifi-p2p #7: Task :react-native-wifi-p2p:compileDebugJavaWithJavac FAILED

SMKH-PRO, 2019-06-04:
Getting this error:- 
I rechecked linking everything is same as defined in documentation. 

Note:-
I have changed the library's gradle version to 3.3.1. and sdks to 28 

```
> Task :react-native-wifi-p2p:compileDebugJavaWithJavac FAILED
Gradle may disable incremental compilation as the following annotation processors are not incremental: databinding-compiler-3.3.1.jar (androidx.databinding:databinding-compiler:3.3.1).
Consider setting the experimental feature flag android.enableSeparateAnnotationProcessing=true in the gradle.properties file to run annotation processing in a separate task and make compilation incremental.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-wifi-p2p:compileDebugJavaWithJavac'.
> javax/xml/bind/JAXBException

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 13s
```

kirillzyusko, 2019-06-04:
@SMKH-PRO Hi, have you tried to add next lines to your `android/build.gradle` file?
```
subprojects {
    afterEvaluate {project ->
        if (project.hasProperty("android")) {
            android {
                    compileSdkVersion 26
                    buildToolsVersion '26.0.2'
            }
        }
    }
}
```

SMKH-PRO, 2019-06-06:
TRIED, Same error stil presist

kirillzyusko, 2019-06-06:
@SMKH-PRO Could you install this library directly from githib and try to build your project?
I've changed `build.gradle` file and pushed these changes recently. I think it will help you.

kirillzyusko, 2019-06-06:
@SMKH-PRO I updated npm-repo (new version is 0.6.6). Now you can intall from there. Also I update repo with examples - you can check it out (https://github.com/kirillzyusko/react-native-wifi-p2p-example). It might be helpful, if you have any questions (now there is also `sdk 28`).

kirillzyusko, 2019-06-16:
@SMKH-PRO Have you resolved your issue? Can I close this topic?

closed 2019-06-21 by kirillzyusko