kirillzyusko/react-native-wifi-p2p #8: TypeError: TypeError: undefined is not an object (evaluating 'WiFiP2PManager.init')

gunaangs, 2019-06-30:
TypeError: TypeError: undefined is not an object (evaluating 'WiFiP2PManager.init')

This error is located at:
    in Share (created by SceneView)
    in SceneView (at StackViewLayout.js:784)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in AnimatedComponent (at StackViewCard.js:69)
    in RCTView (at View.js:44)
    in AnimatedComponent (at screens.native.js:59)
    in Screen (at StackViewCard.js:57)
    in Card (at createPointerEventsContainer.js:27)
    in Container (at StackViewLayout.js:860)
    in RCTView (at View.js:44)
    in ScreenContainer (at StackViewLayout.js:311)
    in RCTView (at View.js:44)
    in AnimatedComponent (at StackViewLayout.js:307)
    in Handler (at StackViewLayout.js:300)
    in StackViewLayout (at withOrientation.js:30)
    in withOrientation (at StackView.js:79)
    in RCTView (at View.js:44)
    in Transitioner (at StackView.js:22)
    in StackView (created by Navigator)
    in Navigator (at createKeyboardAwareNavigator.js:12)
    in KeyboardAwareNavigator (at createAppContainer.js:388)
    in NavigationContainer (at App.js:37)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in Root (at connectStyle.js:384)
    in Styled(Root) (at App.js:33)
    in Provider (at App.js:32)
    in App (at withExpoRoot.js:22)
    in RootErrorBoundary (at withExpoRoot.js:21)
    in ExpoRootComponent (at renderApplication.js:34)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in AppContainer (at renderApplication.js:33)

This error is located at:
    in NavigationContainer (at App.js:37)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in Root (at connectStyle.js:384)
    in Styled(Root) (at App.js:33)
    in Provider (at App.js:32)
    in App (at withExpoRoot.js:22)
    in RootErrorBoundary (at withExpoRoot.js:21)
    in ExpoRootComponent (at renderApplication.js:34)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in AppContainer (at renderApplication.js:33)

Stack trace:
  node_modules\react-native-wifi-p2p\index.js:13:40 in initialize
  src\screens\Share.js:34:4 in componentDidMount
  node_modules\react-native\Libraries\Renderer\oss\ReactNativeRenderer-dev.js:12144:10 in commitLifeCycles
  node_modules\react-native\Libraries\Renderer\oss\ReactNativeRenderer-dev.js:13496:8 in commitAllLifeCycles
  ...

Just I am using your example code and try to run the above error occur.
I am using real device for run the code.

kirillzyusko, 2019-06-30:
Hi @gunaangs 
If you are running my code examples it means that there should not be any errors with library linking (do I correct understand, that you try to run https://github.com/kirillzyusko/react-native-wifi-p2p-example or you copied pieces of code which were given in README.MD from this repository?).
So first of all I'd like to ask you - are you sure that your device support p2p-mode?
You can check it in WiFi settings.

gunaangs, 2019-06-30:
Hi @kirillzyusko 

I have copied the example program and make it use in my app. Just copied App.js and try to run I got this error

kirillzyusko, 2019-06-30:
@gunaangs I got it. I can suppose you were not linking library (follow this guide https://github.com/kirillzyusko/react-native-wifi-p2p/blob/master/README.MD#link-library). Am I right?
In any case you can try to run my example (just clone repository and run it locally). It should work.
And yes, you are right - S7 should support this feature.
I think you have troubles with linking this library.

gunaangs, 2019-06-30:
@kirillzyusko I understood. Actually I am using Expo client to run my application. In that there is no android/settings.gradle file to link those libraries.

Could you help me to solve this issue?

gunaangs, 2019-06-30:
[image]

This is my folder structure

kirillzyusko, 2019-06-30:
@gunaangs here is good explanation: https://stackoverflow.com/questions/44977693/react-native-link-using-expo
Let me know if your issue has been resolved.

kirillzyusko, 2019-06-30:
@gunaangs here is explanation from official Expo docs: https://docs.expo.io/versions/v28.0.0/expokit/detach/#you-might-want-to-detach-if
I think you have to detach your project if you want to use this library.
If you have any other questions, please write me here. I will try to solve them :)

kirillzyusko, 2019-07-08:
@gunaangs Can I close the issue?

closed 2019-07-09 by kirillzyusko

kirillzyusko, 2022-06-21:
Hi @JeanLuisJimenez 

Could you open a new issue and provide more details? When exactly are you getting null is not an object?