kirillzyusko/react-native-wifi-p2p #9: Require Group Passphrase

qalqi, 2019-09-16:
Hi, 
This is a great library. Once a group is created, we need passphrase from it.

Currently, only way to get it is from Settings -> WIFI Direct -> Passphrase

```java
manager.requestGroupInfo(channel, new GroupInfoListener() {
  @Override
  public void onGroupInfoAvailable(WifiP2pGroup group) {
      String groupPassword = group.getPassphrase();
  }
});
```

kirillzyusko, 2019-09-16:
Hi, @qalqi
Yes, you are absolutely right
I will implement this method as soon as possible
Do I correct understand, that you need a method in JS, which will return `groupPassword` from code snippet?

qalqi, 2019-09-16:
Raising PR. Please review it.

kirillzyusko, 2019-09-16:
@qalqi May I ask you why do we need to wait ~3 s in order to call to another function?

qalqi, 2019-09-16:
creategroup.then resulting in null password. If called after a second or so after creating a group, I am receiving group passphrase.

kirillzyusko, 2019-09-16:
It's a little bit strange
But I think we can move forward even with this

closed 2019-09-16 by kirillzyusko

turingpavan, 2019-09-16:
Tx

qalqi, 2019-09-16:
Tx

kirillzyusko, 2019-09-16:
@qalqi I published new version (0.6.7) with your changes :)