kirillzyusko/react-native-wifi-p2p #11: Question: How to work with react-native-multipeer?

qalqi, 2019-09-17:
Can you please illustrate how do we communicate with iOS devices with `react-native-multipee` and Android devices with `react-native-wifi-p2p`?

kirillzyusko, 2019-09-17:
@qalqi Do I correct understand, that you want to connect Android device with iOS using p2p connection?

qalqi, 2019-09-17:
Yep

kirillzyusko, 2019-09-17:
I'm afraid you can't... Because mutipeer connectivity cannot connect iOS and Android devices. It's not a react-native restriction - it's a platform restriction (since they are used different approach in realisation) :(
You need to write your own implementation I think

qalqi, 2019-09-17:
I was able to connect a Windows sdevice, a macbook device and a iOS device to WIFI direct group created by `createGroup` 
Only thing I need now, at least in iOS device to be able to send a message

kirillzyusko, 2019-09-17:
Hm, have you tried to call `send` function from `react-native-multipeer`?
Or call `sendMessage` on Android side in order to send some data to iOS devices?

qalqi, 2019-09-18:
iOS to iOS, peer advertising and connecting is working fine.
Create a WIFI Direct group from one Android device and able to connect to it with WIFI Direct group passphrase then create a Multipeer iOS to iOS connection.

Now, Android to iOS is something I haven't been able to figure out.

kirillzyusko, 2019-09-18:
I think, I can't help you with this issue, unfortunately.
If you find a solution, please, write it here

kirillzyusko, 2019-09-19:
@qalqi And also you can have a look on https://developers.google.com/nearby/

kirillzyusko, 2019-09-26:
@qalqi I close this issue. If you find something interesting, please leave your comment here. Thank you in advance.

closed 2019-09-26 by kirillzyusko