kirillzyusko/react-native-wifi-p2p #13: startDiscoveringPeers returns undefined

anthonyhungnguyen, 2019-11-01:
[image]

I tried to follow all steps, but seems like I cannot get any peers around because of undefined return when discovering peers. Any help ?

kirillzyusko, 2019-11-01:
@anthonyhungnguyen Thank you for the issue.
It is ok, that you can see undefined status :)
Could you please answer on several questions?
- What is the version of library and RN do you use?
- What is the order of calling API methods of this library?
- Do you see others devices with P2P mode on your device from Settings?
- Do you run it on real device?

anthonyhungnguyen, 2019-11-02:
Oh, okay:
1. 
[image]
2. 
[image]
3. From setting, using wifi direct, I can see others devices.
[image]
4. Yes, I ran it on HTC U Ultra (version 8.0.0) and Samsung Tablet SM-T231 ( version 4.4.2)

anthonyhungnguyen, 2019-11-02:
[image]
[image]
I don't know why getting no peers even 2 devices on the same wifi
I ran 2 different server on two ports: 5000 and 5050 using `react-native start --port <port_number>`
Please help me

kirillzyusko, 2019-11-02:
@anthonyhungnguyen Can you try just enable p2p mode on one of your device (on android 4.4.2) and run your React Native application on the second (android 8.0)
And start discovering from 8.0 in RN application?
If it doesn't work, you can try change roles of them. I mean just enable wifi p2p mode on 8.0 and start discovering from 4.x.x
Also as an option you can try to create a group on one of device and try to detect this group from another (this step you can do from installed application on both devices)
I also faced with the same problem, but in my case I can see android 8.0 from android 4.2 but not vice versa (but in settings I could see everything) :(
As a last option - could you collect more devices and try to run on them?

tmxkwkfgka, 2019-11-06:
I'm having same problem also.

when I searched available device from galaxy s7 (android version 8.0). It returned empty array. but when I searched available device from galaxy note4 (andorid version 6.0.1). It returned available device. I should make it work on android 8.0. Do you know why It is not work on specific version and can you fix it?

Thank you.

tmxkwkfgka, 2019-11-06:
`11-06 14:14:20.440: D/WifiHandler.WifiP2pService(3663): Received message=139412d sendingUid=10389
11-06 14:14:20.440: D/WifiHandler.WifiP2pService(3663): Received message=139265d sendingUid=10389
11-06 14:14:20.441: D/WifiP2pService(3663): InactiveState{ when=-1ms what=139412 obj=Bundle[mParcelledData.dataSize=204] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.442: D/WifiP2pService(3663): P2pEnabledState{ when=-1ms what=139412 obj=Bundle[mParcelledData.dataSize=204] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.443: D/WifiP2pService(3663): DefaultState{ when=-2ms what=139412 obj=Bundle[mParcelledData.dataSize=204] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.444: D/WifiP2pService(3663): InactiveState{ when=-2ms what=139265 arg2=32 target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.445: D/WifiP2pService(3663): P2pEnabledState{ when=-3ms what=139265 arg2=32 target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.445: D/WifiP2pService(3663): P2pEnabledState package to call DISCOVER_PEERS (channel : 0) -> com.wifi_d
11-06 14:14:20.459: D/WifiAsyncChannel.WifiP2pService(3663): replyToMessage recvdMessage=139265 sendingUid=10389 sentMessage=139267
11-06 14:14:20.487: D/wifip2pmanager(31404): before request peers
11-06 14:14:20.489: D/WifiHandler.WifiP2pService(3663): Received message=139283d sendingUid=10389
11-06 14:14:20.490: D/WifiP2pService(3663): InactiveState{ when=0 what=139283 arg2=33 obj=Bundle[mParcelledData.dataSize=116] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.491: D/WifiP2pService(3663): P2pEnabledState{ when=-1ms what=139283 arg2=33 obj=Bundle[mParcelledData.dataSize=116] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.494: D/WifiP2pService(3663): DefaultState{ when=-2ms what=139283 arg2=33 obj=Bundle[mParcelledData.dataSize=116] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:20.495: D/WifiAsyncChannel.WifiP2pService(3663): replyToMessage recvdMessage=139283 sendingUid=10389 sentMessage=139284
11-06 14:14:20.496: D/wifip2pmanager(31404): available devices 
11-06 14:14:22.108: D/WifiP2pService(3663): InactiveState{ when=-1ms what=147477 obj=
11-06 14:14:22.108: D/WifiP2pService(3663):  deviceAddress: 32:c7:1c
11-06 14:14:22.108: D/WifiP2pService(3663):  primary type: 00030050F2040005
11-06 14:14:22.108: D/WifiP2pService(3663):  secondary type: null
11-06 14:14:22.108: D/WifiP2pService(3663):  wps: 136
11-06 14:14:22.108: D/WifiP2pService(3663):  grpcapab: 9
11-06 14:14:22.108: D/WifiP2pService(3663):  devcapab: 4
11-06 14:14:22.108: D/WifiP2pService(3663):  status: 3
11-06 14:14:22.108: D/WifiP2pService(3663):  wfdInfo: WFD enabled: trueWFD DeviceInfo: 0
11-06 14:14:22.108: D/WifiP2pService(3663):  WFD CtrlPort: 0
11-06 14:14:22.108: D/WifiP2pService(3663):  WFD MaxThroughput: 0
11-06 14:14:22.108: D/WifiP2pService(3663):  groupownerAddress: null
11-06 14:14:22.108: D/WifiP2pService(3663):  GOdeviceName: null
11-06 14:14:22.108: D/WifiP2pService(3663):  interfaceAddress: 
11-06 14:14:22.108: D/WifiP2pService(3663):  SConnectInfo : null
11-06 14:14:22.108: D/WifiP2pService(3663):  contactInfoHash : null
11-06 14:14:22.108: D/WifiP2pService(3663):  ssDevInfo : 0
11-06 14:14:22.108: D/WifiP2pService(3663):  iconIdx : 0
11-06 14:14:22.108: D/WifiP2pService(3663):  semSamsungDeviceType : 0
11-06 14:14:22.108: D/WifiP2pService(3663):  serviceData : null
11-06 14:14:22.108: D/WifiP2pService(3663):  fw_invite : 0 target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:22.109: D/WifiP2pService(3663): P2pEnabledState{ when=-3ms what=147477 obj=
11-06 14:14:22.109: D/WifiP2pService(3663):  deviceAddress: 32:c7:1c
11-06 14:14:22.109: D/WifiP2pService(3663):  primary type: 00030050F2040005
11-06 14:14:22.109: D/WifiP2pService(3663):  secondary type: null
11-06 14:14:22.109: D/WifiP2pService(3663):  wps: 136
11-06 14:14:22.109: D/WifiP2pService(3663):  grpcapab: 9
11-06 14:14:22.109: D/WifiP2pService(3663):  devcapab: 4
11-06 14:14:22.109: D/WifiP2pService(3663):  status: 3
11-06 14:14:22.109: D/WifiP2pService(3663):  wfdInfo: WFD enabled: trueWFD DeviceInfo: 0
11-06 14:14:22.109: D/WifiP2pService(3663):  WFD CtrlPort: 0
11-06 14:14:22.109: D/WifiP2pService(3663):  WFD MaxThroughput: 0
11-06 14:14:22.109: D/WifiP2pService(3663):  groupownerAddress: null
11-06 14:14:22.109: D/WifiP2pService(3663):  GOdeviceName: null
11-06 14:14:22.109: D/WifiP2pService(3663):  interfaceAddress: 
11-06 14:14:22.109: D/WifiP2pService(3663):  SConnectInfo : null
11-06 14:14:22.109: D/WifiP2pService(3663):  contactInfoHash : null
11-06 14:14:22.109: D/WifiP2pService(3663):  ssDevInfo : 0
11-06 14:14:22.109: D/WifiP2pService(3663):  iconIdx : 0
11-06 14:14:22.109: D/WifiP2pService(3663):  semSamsungDeviceType : 0
11-06 14:14:22.109: D/WifiP2pService(3663):  serviceData : null
11-06 14:14:22.109: D/WifiP2pService(3663):  fw_invite : 0 target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:22.117: D/WifiHandler.WifiP2pService(3663): Received message=139283d sendingUid=1000
11-06 14:14:22.121: D/WifiP2pService(3663): InactiveState{ when=-1ms what=139283 arg2=309 obj=Bundle[{android.net.wifi.p2p.CALLING_PACKAGE=android}] target=com.android.internal.util.StateMachine$SmHandler }
[... cut ...]
11-06 14:14:23.691: D/WifiP2pService(3663): P2pEnabledState{ when=-2ms what=139283 arg2=35 obj=Bundle[mParcelledData.dataSize=116] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:23.692: D/WifiP2pService(3663): DefaultState{ when=-3ms what=139283 arg2=35 obj=Bundle[mParcelledData.dataSize=116] target=com.android.internal.util.StateMachine$SmHandler }
11-06 14:14:23.694: D/WifiAsyncChannel.WifiP2pService(3663): replyToMessage recvdMessage=139283 sendingUid=10389 sentMessage=139284
`

This is andorid log when I searched available device on android version 8. Maybe It can detect available device. but can't return device info to react native code.

kirillzyusko, 2019-11-06:
@tmxkwkfgka hi! Could you please try to ask `ACCESS_COARSE_LOCATION` permission dynamically (on runtime) on Android 8 before wi-fi usage?
Similar issue: https://stackoverflow.com/questions/50475371/android-wifi-direct-p2p-not-able-to-find-peers-on-android-8
And as I understand it's Android version specific problem. Not a bug of this library.
Could you please try to ask permissions, or try anything else from stackoverflow thread and leave your comment here about whether it helped ot not?

tmxkwkfgka, 2019-11-06:
Thank you. It works.
I modiried WiFiP2PManagerModule.java
`@ReactMethod
    public void getAvailablePeersList(final Promise promise) {
        Activity activity = getCurrentActivity();

        if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.M && ContextCompat.checkSelfPermission(this.reactContext, Manifest.permission.ACCESS_COARSE_LOCATION) != PackageManager.PERMISSION_GRANTED){
            ActivityCompat.requestPermissions(activity, new String[]{Manifest.permission.ACCESS_COARSE_LOCATION},
                            PERMISSIONS_REQUEST_CODE_ACCESS_COARSE_LOCATION);
            //After this point you wait for callback in onRequestPermissionsResult(int, String[], int[]) overriden method

        }else{
            Log.d(logTag, "before request peers");
            manager.requestPeers(channel, new PeerListListener() {
                @Override
                public void onPeersAvailable(WifiP2pDeviceList deviceList) {
                    Log.d(logTag, "available devices "+ deviceList.getDeviceList().size());
                    WritableMap params = mapper.mapDevicesInfoToReactEntity(deviceList);
                    promise.resolve(params);
                }
            });
        }
        
    }
`

kirillzyusko, 2019-11-06:
@tmxkwkfgka Wow! Thank you for the response!
Unfortunattely I haven't found `PERMISSIONS_REQUEST_CODE_ACCESS_COARSE_LOCATION` permission in Android from `react-native`.
I will release new version as soon as possible and there I will add new method to API for checking this permission.

anthonyhungnguyen, 2019-11-07:
I figure it out, follow @tmxkwkfgka and google on [https://www.javatips.net/api/Lamp-master/app/src/main/java/com/lamps/lamps/activity/WifiConnectActivity.java](url)

This is my WiFiP2PManagerModule.java
Adding more imports:

`import android.os.Build;
import android.Manifest;
import androidx.core.content.ContextCompat;
import androidx.core.app.ActivityCompat;`

`@ReactMethod
    public void getAvailablePeersList(final Promise promise) {
        Activity activity = getCurrentActivity();
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M && ContextCompat.checkSelfPermission(this.reactContext,
                Manifest.permission.ACCESS_COARSE_LOCATION) != PackageManager.PERMISSION_GRANTED) {
            ActivityCompat.requestPermissions(activity, new String[] { Manifest.permission.ACCESS_COARSE_LOCATION },
                    PERMISSIONS_REQUEST_CODE_ACCESS_COARSE_LOCATION);
            // After this point you wait for callback in onRequestPermissionsResult(int,
            // String[], int[]) overriden method

        } else {
            manager.requestPeers(channel, new PeerListListener() {
                @Override
                public void onPeersAvailable(WifiP2pDeviceList deviceList) {
                    WritableMap params = mapper.mapDevicesInfoToReactEntity(deviceList);
                    promise.resolve(params);
                }
            });
        }

    }`

Then it works

kirillzyusko, 2019-11-07:
@anthonyhungnguyen Yes, sure
But look at if-else statement.
Do I correct understand, that if you call this method first time without permisiions granting, then we just will ask about permissions, but after granting we will not call `requestPeers` in else statement?
I think it is incorrect, since you need first time call this method and after granting permissions also call it second time?
Am I right?

kirillzyusko, 2019-11-07:
@anthonyhungnguyen I've just published new version without Java-code modification.
Could you give a shot and try to use new JS version of `onStartInvestigate ` method from examples in documentation?
You can find it here: https://github.com/kirillzyusko/react-native-wifi-p2p#example-of-usage
And let me know, if your problem was resolved. Also, please, delete your old version of an application (in this case old permissions will be deleted, and I can assure, that permission was granted successfully via JS methods without modifying Java-code)

anthonyhungnguyen, 2019-11-07:
I have tried, it works perfectly, thanks alot
But when I test it with sendMessage, I got this warning

[image]
The image above is the device I use to sendMessage('hello'), but when I switch to my samsung tablet console, nothing receive.
[image]

I wonder whether I implement it incorrectly
[image]

kirillzyusko, 2019-11-07:
@anthonyhungnguyen Did you complete the following steps?

https://github.com/kirillzyusko/react-native-wifi-p2p#sendmessagemessage

closed 2019-11-07 by kirillzyusko

muzammil413tariq, 2021-04-29:
Hey I wants an app which can send data from my react native mobile app to my desktop please guide me how can i achieve this

kirillzyusko, 2021-04-29:
@muzammil413tariq could you please create a new issue?