kirillzyusko/react-native-wifi-p2p #15: send Message is not working

vamsig7, 2020-02-21:
when i tried to use send message
I am gettting error "Attemp to invoke virtual method ' java.lang.String java.net.InetAdress.getHostAdress()' on anull object reference

kirillzyusko, 2020-02-21:
Hi
It seems, like here is an Exception:
```
wifiP2pInfo.groupOwnerAddress.getHostAddress()
```
I think `groupOwnerAddress==null`
Did you follow these steps: https://github.com/kirillzyusko/react-native-wifi-p2p#sendmessagemessage ?

vamsig7, 2020-02-21:
Thanx for immediate reply
I am pasting my results step wise 
1.)log of create Group on server side  Group created successfully!
2.)I connceted to that Group(same name as in Wifi Direct)
3.) got the getCOnnectionInfo at the client  end {isGroupOwner:false,groupFOrmed:true:groupaddress {{}}}
that group address column has an ip adress some thinng like 192.xxx.xxx
4.when use sendMessage after all this i got that error

Note:when I tried to get the connection Info in server side its showing groupowner as false and group for med as false though createGroup was sucessfull

kirillzyusko, 2020-02-21:
After calling `getConnectionInfo` - do you see `hostAddress` field?

vamsig7, 2020-02-21:
{"groupFormed": false, "groupOwnerAddress": null, "isGroupOwner": false}
ouput on server side .
{isGroupOwner: false, groupFormed: true, groupOwnerAddress: {…}}
isGroupOwner: false
groupFormed: true
groupOwnerAddress: {isLoopbackAddress: false, hostAddress: "192.168.49.1"}
on receiver side

kirillzyusko, 2020-02-21:
Could you please try to send a message from receiver side?

vamsig7, 2020-02-21:
I tried to log values with some delay on server side
{"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": true}
but same result  ntng changed

kirillzyusko, 2020-02-21:
Have you tried to send a message from server to client?
I see, that `hostAddress` is not null on server side => you shouldn't see error, that you described above.
Do you still see it, even with sending a message from server to client?

vamsig7, 2020-02-21:
Yes I got the same error on the receiver side  i.e server to client

kirillzyusko, 2020-02-21:
It's weird!
What can I suggest - try tomodify native Java code (merely add `System.out.println`) and via `adb logcat` see, how varibles are changing.
I assume, there might be some kind of race condition... However, on my devices everything worked fine. I coud send messages/files.
I think I can not help you directly, since I don't have an access to your devices.
Just look at these lines: 
1 - https://github.com/kirillzyusko/react-native-wifi-p2p/blob/master/android/src/main/java/io/wifi/p2p/WiFiP2PManagerModule.java#L61 - here you receive an object with `hostAddress` and save it in class variable
2 - https://github.com/kirillzyusko/react-native-wifi-p2p/blob/master/android/src/main/java/io/wifi/p2p/WiFiP2PManagerModule.java#L258 - here you use this saved object (but it's null and you get an error)
3 - https://github.com/kirillzyusko/react-native-wifi-p2p/blob/master/android/src/main/java/io/wifi/p2p/WiFiP2PManagerModule.java#L51 - as an option look here. Maybe it will be a cause of a race condition
I hope I helped you a little bit :)
I'm so sorry. I really would like to help you, but I don't have second device - so I can't.

vamsig7, 2020-02-21:
ok fine Will try to work aroung with Java and update here if it works

kirillzyusko, 2020-03-20:
@vamsig7 any updates? :)

kirillzyusko, 2020-04-05:
Closed due to inactivity.

closed 2020-04-05 by kirillzyusko