kirillzyusko/react-native-wifi-p2p #16: Why is getConnectionInfo() marked temporary?

NoamDev, 2020-04-14:


kirillzyusko, 2020-04-14:
Hi, @NoamDev 
Well, it's good question :)
If you look at implementation of this method (https://github.com/kirillzyusko/react-native-wifi-p2p/blob/master/android/src/main/java/io/wifi/p2p/WiFiP2PManagerModule.java#L61) you can see this function has side effect (saving new info into local `wifiP2pInfo` variable).
And this method is only used in sendMessage/sendFile flow. And in future I would like to simplify these flows and get rid off the usage of this method.
That's why I marked this method as `temporary` - I don't know whether we need to have this method in future after simplifying flow or not :)
Nonetheless if I make breaking change I will publish new version as 2.x.x and prepare some migration documents.
So now you can use this method.
Did I answer on your question? :)

NoamDev, 2020-04-14:
Hi, thanks for quick reply.
May I ask what's bad with "saving new info into local wifiP2pInfo variable"?
Be aware that I, for example, do not intend to use the sendMessage
function. I plan to use basic internet protocols (in that case for webrtc).

kirillzyusko, 2020-04-14:
The lifecycle method is not an evil 😅 
I was just curios why did I need to use this method in the sending messages/files flow. Why did I need to update a variable? As far as I remember I didn't see anything in official docs - that's why I marked this method as `temporary`.
I think I should remove this label :)

NoamDev, 2020-04-15:
Oh, I didn't notice that this is a totally useless line of code😂

>
Yeah, you should probably get rid of it..

P.s. I'm just curious, what docs are talking about?

closed 2020-04-16 by NoamDev