kirillzyusko/react-native-wifi-p2p #17: startDiscoveringPeers() always rejects with code 0 (ERROR)

NoamDev, 2020-04-16:
Yes, I request the ACCESS_FINE_LOCATION permission.
What else could cause that?
thanks.

kirillzyusko, 2020-04-16:
Hi, @NoamDev
It might be helpful: https://github.com/kirillzyusko/react-native-wifi-p2p/issues/13
After calling `startDiscoveringPeers()` you always get and error in `catch`?

NoamDev, 2020-04-16:
Yes, it always results with a catch.
Thats my code:
```
    wifiP2P.initialize();
    const status = await wifiP2P.isSuccessfulInitialize();
    if (!status) {
      console.error('wifip2p initialization was not successful');
    } else {
      console.log('wifip2p initialized');
    }
    await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
    );
    try {
      await wifiP2P.startDiscoveringPeers();
      console.log('Discovering peers started');
    } catch (e) {
      console.log('Could not start peers discovery');
      console.log(e.code);
    }
````
dependecies:
```
"dependencies": {
    "react": "16.11.0",
    "react-native": "0.62.2",
    "react-native-wifi-p2p": "^1.0.2"
  },
```

kirillzyusko, 2020-04-16:
Does your wifi turned on?
And phone supports p2p mode?

kirillzyusko, 2020-04-16:
Also did you try to run on different device?
I can't say anythin about your case, since you receive general error. You can read about this: https://developer.android.com/reference/android/net/wifi/p2p/WifiP2pManager#ERROR
I think it's not an issue of the library.

NoamDev, 2020-04-16:
it worked on another device, thank you!

NoamDev, 2020-04-16:
OH! It turned out that on new android versions the location should be on too!
Now it works on the pixel as well.

NoamDev, 2020-04-16:
Thank for your time, you've got my star:)

closed 2020-04-16 by NoamDev

kirillzyusko, 2020-04-16:
> It turned out that on new android versions the location should be on too!
Nice to know :)
Maybe should we change README file? In order to avoid duplicate issues in future?
Could you provide your working code, please?

NoamDev, 2020-04-17:
The only things I've changed is:

1. "COARSE_LOCATION"  permission --> "FINE_LOCATION" permission (since android Q, but it'll work for prevoius versions too.)

2. both wifi and location should be enabled, but that has nothing to do with my code (although I'll probably use something like [react-native-android-location-enabler](https://github.com/Richou/react-native-android-location-enabler)). this is required since android oreo according to [a stackoverflow's answer](https://stackoverflow.com/a/53149375/8214617)