kirillzyusko/react-native-wifi-p2p #18: Group creation is not defined on real-time

DarkDracoon, 2020-04-21:
Device A:
- Device: Android
- OS: API 29 (Q)

Device B:
- Device: Android
- OS: API 28 (P)

Using the master version of react-native-wifi-p2p or the project itself for demonstration of the library API, I'm not able to represent a group creation in real-time on the opposite device under these two devices, by getting the available devices, even though it is available on connection info. Is this a known issue for the newer Android devices or so? Without this, I can't either connect.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the example app
2. Create Group
3. On the guest (client that will connect to the group owner), press Get Available Devices button, and you'll get "isGroupOwner": false.

kirillzyusko, 2020-04-21:
Hi @DarkDracoon 
Are you able to get available peers/discover them?

DarkDracoon, 2020-04-23:
Yes @kirillzyusko, although the "isGroupOwner" has a really large delay on showing as a group owner.

kirillzyusko, 2020-04-24:
@DarkDracoon Did you request COARSE_LOCATION and FINE_LOCATION permissions?
Also you need turn on GPS/Location service: https://stackoverflow.com/questions/50475371/android-wifi-direct-p2p-not-able-to-find-peers-on-android-8/53149375#53149375

DarkDracoon, 2020-04-26:
Yes @kirillzyusko , I did turn on Location and requested the specific permissions. I may debug that tomorrow.

kirillzyusko, 2020-04-26:
@DarkDracoon I'm looking forward to your results😎

kirillzyusko, 2020-05-02:
@DarkDracoon Hi, any updates?

kirillzyusko, 2020-05-06:
Closed due to inactivity.

closed 2020-05-06 by kirillzyusko