kirillzyusko/react-native-wifi-p2p #19: a proposal: use firebase  testlab

NoamDev, 2020-04-21:
What do you think about inserting some CI to your development?
You can first test using mocks, but then you can test using real devices with firebase testing lab.
It won't help you to connect to actual devices probably, but you _can_ check things like initializing discover peers. This way, you'd be able to generally test on new devices and android versions you do not possess.
it would have prevented my issue for examaple:)

kirillzyusko, 2020-04-21:
Hi @NoamDev 
It sounds like an awesome idea😎
Unfortunately, right now I don't have enough time for implementing this CI pipeline😒
But you can shoot a MR - it's highly appreciate ☺️

NoamDev, 2020-04-22:
Pardon? MR?

kirillzyusko, 2020-04-22:
Ah, sorry.
In different VCS systems it's called different. So, in GitLab it's MR (merge request), in GitHub it's called PR (pull request). When I wrote MR I meant PR :)

kirillzyusko, 2020-04-29:
@NoamDev I hope I see PR soon from you😄
In any case, in a couple of months I’ll try to add support myself - now I can’t, because I’m very busy😞
Thank you for the suggestion😎

closed 2020-04-29 by kirillzyusko

NoamDev, 2020-05-06:
Just to note, it turns out firebase testing lab does not support wifi
direct, so such idea, would have to be implemented using another provider.

kirillzyusko, 2020-05-06:
Do you know any free providers, which give an access to wi-fi?

NoamDev, 2020-05-08:
A quick google search suggested [kobiton](https://kobiton.com/).
Free plan includes 120 min/month and 50 devices.
And yes, they support wifi.
Worth a try I believe, but I too, don't have time for this currently.

kirillzyusko, 2020-05-08:
@NoamDev Got it. Thank you for research.
Maybe after 1-2months I can try to integrate it :)
In any case - thank you for your time and idea! 😎