kirillzyusko/react-native-wifi-p2p #20: disconnect implementation doesn't match the name 'disconnect'

NoamDev, 2020-04-29:
currently the implementation is:
```
@ReactMethod
    public void disconnect(final Callback callback) {
        manager.cancelConnect(channel, new WifiP2pManager.ActionListener() {
            @Override
            public void onSuccess() {
                callback.invoke();
            }

            @Override
            public void onFailure(int reasonCode) {
                callback.invoke(Integer.valueOf(reasonCode));
            }
        });
    }
```

However, according to [its docs](https://developer.android.com/reference/android/net/wifi/p2p/WifiP2pManager#cancelConnect(android.net.wifi.p2p.WifiP2pManager.Channel,%20android.net.wifi.p2p.WifiP2pManager.ActionListener)) cancelConnect - 'Cancel any ongoing p2p group negotiation'.

disconnect should be like in this [stackoverflow answer](https://stackoverflow.com/a/18792707/8214617):

```
public static void disconnect() {
    if (mManager != null && mChannel != null) {
        mManager.requestGroupInfo(mChannel, new GroupInfoListener() {
            @Override
            public void onGroupInfoAvailable(WifiP2pGroup group) {
                if (group != null && mManager != null && mChannel != null) {
                    mManager.removeGroup(mChannel, new ActionListener() {

                        @Override
                        public void onSuccess() {
                            Log.d(TAG, "removeGroup onSuccess -");
                        }

                        @Override
                        public void onFailure(int reason) {
                            Log.d(TAG, "removeGroup onFailure -" + reason);
                        }
                    });
                }
            }
        });
    }
}
```

NoamDev, 2020-04-29:
I don't what is wrong, the name 'disconnect' or the implementation.
@kirillzyusko, What was your purpose?

kirillzyusko, 2020-04-29:
Hi, @NoamDev 
When I worked on implementation of this library, I merely wanted to recreate all basics methods (yes, sometimes methods names are different :smile:)

About `disconnect` implementation from StackOverflow - you can re-create this code snippet using JS code:

```
await getGroupPassphraseInfo()
await removeGroup()
```

It was a goal of this library: don't try to cover all possible cases. Just write basics methods and if needed - combine them in JS

NoamDev, 2020-04-29:
Yeah, I guess that what enabled to create a great library with only two collaborators:)
In that case, I suggest you to rename the method, it is really misleading..

NoamDev, 2020-04-29:
Yes, I know, it's a breaking change so you'd have to publish a new version...

kirillzyusko, 2020-04-29:
Yes, I agree with you
I definetly have to do it
Thank you for pointing it out :)

closed 2020-04-29 by kirillzyusko

kirillzyusko, 2020-04-29:
@NoamDev Done!😎 I published new version as `2.0.0`. So, you can update to the newest and use `cancelConnect`.