kirillzyusko/react-native-wifi-p2p #21: [feature request] requestGroupInfo equivalent

NoamDev, 2020-04-30:
Currently, there's a function called getGroupPassphraseInfo, but it'll only give you the passphrase.
I need to check who are the other peers in my group.
I need the groupOwner's and clients' macAddresses.

I believe the best way to achieve this, is to create additional method requestGroupInfo:
```
    @ReactMethod
    public void getGroupInfo(final Promise promise) {
        manager.requestGroupInfo(channel, new WifiP2pManager.GroupInfoListener() {
            @Override
            public void onGroupInfoAvailable(WifiP2pGroup group) {
                promise.resolve(group);
            }
        });
    }

```

and an appropriate js method:
`const getGroupInfo = () => WiFiP2PManager.getGroupInfo();`

Maybe you should also deprecate the getGroupPassphraseInfo method and remove it in the next breaking change.

NoamDev, 2020-04-30:
If you prefer that, I can send a PR:)

NoamDev, 2020-04-30:
Oh, I'm sorry, It's not that simple.
I should also create a mapping method.
would you like me to send a PR?

kirillzyusko, 2020-04-30:
@NoamDev Yes, you are right, it's not so simple and you also need to create a mapper :)
That would be really nice, if you can create a MR and also test your changes 😎

kirillzyusko, 2020-04-30:
@NoamDev Also, have you thought about implementation `GroupInfoListener` in `WiFiP2PBroadcastReceiver`?
If peers in group change, will you receive an action/intent in `onReceive`?
If yes, I would suggest you implement new listener, and emit JS events, if info about peers in group has changed.
What do you think about it?
Because, as I could understand, if you implement `getGroupInfo` - you will call it via `setInterval`, in order to get every time actual information. Am I correct?

kirillzyusko, 2020-04-30:
@NoamDev I think, implementation should be similar to https://stackoverflow.com/a/25714357/9272042

NoamDev, 2020-04-30:
I can't find an action that will be triggered whenever a new client joins..
Maybe, maybe, connection changed but the documentation does not imply such thing.

kirillzyusko, 2020-04-30:
@NoamDev that's weird
I think `onReceive` should be triggered
Can you please check it? Just log `action` variable? Using System.out.println or Log.i?

NoamDev, 2020-04-30:
I'd

NoamDev, 2020-05-01:
Ah! You seem to be right!
WifiP2p is one of the least documented APIs...
Anyway, according to this answer: https://stackoverflow.com/a/44119574/8214617
There's an extra for group info which comes with connection_changed action. however, it is only since api level 18.
Do you think it implies the action will be triggered even before api level 18?
Obviously, we wouldn't want to limit the library to api level 18.

kirillzyusko, 2020-05-01:
@NoamDev It seems to be thruth - `EXTRA_WIFI_P2P_GROUP` is only available since API >= 18.
And definetly, I wouldn't like to restict usage of library only because of one method.
I would suggest implement additional logic for mapping new info in `WIFI_P2P_CONNECTION_CHANGED_ACTION ` handler.

And for old API (<= 17) create direct method `getGroupInfo`, and mark it in README smth like "according to official documentation, you should use it, only if you support API <= 17".

What do you think about it?

BTW, what is the action do you receive (on group owner side), when any device connects to group? `WIFI_P2P_CONNECTION_CHANGED_ACTION ` or `WIFI_P2P_THIS_DEVICE_CHANGED_ACTION`?

NoamDev, 2020-05-05:
> And for old API (<= 17) create direct method getGroupInfo, and mark it in README smth like "according to official documentation, you should use it, only if you support API <= 17".

It ain't depreacted or something, I don't see a reason to document that way. Maybe more like, in api>=18, you can also get it through subscribeOnConnectionChanged.

> BTW, what is the action do you receive (on group owner side), when any device connects to group? WIFI_P2P_CONNECTION_CHANGED_ACTION or WIFI_P2P_THIS_DEVICE_CHANGED_ACTION?

What;s exactly the case, after `createGroup`+`connect `/ `connect ` to a device/ `connect` to a group with already 2 clients. Also, I don't have a pre api 18 phone, and with api 18 it is pretty obvious WIFI_P2P_CONNECTION_CHANGED_ACTION  would be triggered.

kirillzyusko, 2020-05-05:
Cool, then I'm awaiting PR from you :blush:

kirillzyusko, 2020-05-14:
@NoamDev any updates? :)

NoamDev, 2020-05-14:
I'm sorry, I don't think I'm gonna send one.
As you probably know, people get to know open source projects when developing a project of their own. However , I've lost interest in that project and so, I don't think I'm gonna see react native anytime soon.

I Hope I did help a bit, and that you or someone else will continue making this tiny yet wonderful library.
Sorry, I won't be able to join and good luck!

closed 2020-05-14 by kirillzyusko