kirillzyusko/react-native-wifi-p2p #22: Sendfile is not working

erdeepsangani, 2020-05-28:
Sendfile is not working it log error

kirillzyusko, 2020-05-28:
Hi @erdeepsangani 
What is the error?

erdeepsangani, 2020-05-28:
Sir in device  array secondary device is null

kirillzyusko, 2020-05-28:
In which array?
Could you attach log from JS + native logs?

erdeepsangani, 2020-05-29:
sir what is a syntex of uri to send a file from internal storage
 for android device

erdeepsangani, 2020-05-29:
sir when i send file function called this error in occers:-
  Error while file sending {"code": "soon will be", "message": "Unknown error."}

kirillzyusko, 2020-05-29:
@erdeepsangani could you capture `adb logcat` logs?

> sir what is a syntex of uri to send a file from internal storage

These URL's were working :)

```
const url = '/storage/sdcard0/Music/Rammstein:Amerika.mp3';
const url = '/storage/emulated/0/Music/Bullet For My Valentine:Letting You Go.mp3';
```

erdeepsangani, 2020-05-30:
I am using react-native-wifi-p2p in one of my project. as seen in logs and Wifi Direct network I am able to find Discover Devices and connect one of them. But when I try to send the message or file In Log Shows the Error "Error while message sending {"code": "soon will be", "message": "Unknown error."}"
I am using the same code given in repo for testing. So please help me to resolve it.

Log of connect is
"64646776467 {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": false}"

Log of getConnectionInfo() is
"{"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": false}"

erdeepsangani, 2020-05-30:
sir what should i do??????????????

erdeepsangani, 2020-05-30:
please sir answer me

kirillzyusko, 2020-05-30:
I am asking about native/java logs. Not logs from JS. Internally in library happens unhandled exception. And I can not send it to JS, since it's unexpected. That's why I am asking about Java logs captured via `adb logcat`.

erdeepsangani, 2020-06-03:
Sir I am able to send only jpg file. When I am trying to send mp4 file.. It receive to receiver phone but this file is currupted and it's extension is. Jpg.....
Sir what am I do now??

kirillzyusko, 2020-06-03:
It seems to be truth...
I hardcoded extension and filename on receiver side.
As an option I can suggest you next combination:
- sendMessage with correct filename
- sendFile
- on receiver side using react-native-fs rename it to correct filename
Yes, it sounds like a hack and I definitely need to fix this issue in my library. But I can not say how soon it happens...

erdeepsangani, 2020-06-03:
Sir please inform me when ever you solve the issue.. Please sir... My mail ID is. Drsangani1210@gmail.com please sir..

kirillzyusko, 2020-06-03:
I would say that optimistic term is July.
But I will try to keep you informed.

fncallback, 2020-06-20:
@kirillzyusko @erdeepsangani  I'm using this library and it is working perfectly fine. I was able to share any file across the devices.

kirillzyusko, 2020-06-20:
@fncallback Hi. Nice to know :)
How did you solve the problem with file name and file extension? As far as I remember I hard coded it. So any received file has `.jpg` as exension and the same file name (or timestamp - I don't remember exactly how I implemented it).

fncallback, 2020-06-21:
@kirillzyusko yes made change in code to send file with any extension

kirillzyusko, 2020-06-21:
@fncallback Nice😎
PRs are highly appreciate😉☺️

erdeepsangani, 2020-06-22:
@fncallback  sir can you write how you make it happen

sandipchandra17, 2020-07-04:
Dear sir, I am facing the same issue. I changed the file name and path according to the host(server) mobile like /storage/emulated/0/docs/filename.extension format but geeting the same above error.

kirillzyusko, 2020-07-05:
Hi @sandipchandra17
File path should start with `file://` prefix

kirillzyusko, 2020-07-05:
Since we clarified, that method is working I'm going to close this issue.
For all other problems, which were discovered here I will create separated issues.

closed 2020-07-05 by kirillzyusko

sandipchandra17, 2020-07-05:
> @kirillzyusko yes made change in code to send file with any extension

Dear sir,  can you please send an example code where did you change the file name and extension.. as file is receving properly in my case but extensiin always is .jpg for all kind of files and fulename is wifip2pshared -timeatamp and in gallery actual jpg file not displayed.

sandipchandra17, 2020-07-05:
> Hi @sandipchandra17
> File path should start with `file://` prefix

This is solved but why error message is throwing like error while file receiving code will be soon.. as per ur code its not showing succesfull msg rather going to catch block though file received in receiver mobile properly.

kirillzyusko, 2020-07-05:
I'm going to fix the problem with extension/filename (you created new issue about this problem, so I'm going to post all actual info there).

According to another problem (why error is throwing in case of successful transfer) - could you please create new issue and attach actual logcat logs there?

P.S. This thread is too long, so I wouldn't recommend post any info here, since it's really difficult to track issues for other people who is trying to search their problem.

kirillzyusko, 2020-07-06:
@sandipchandra17 @fncallback @erdeepsangani @sharmakumaraditya I fixed this issue in `3.0.0`