kirillzyusko/react-native-wifi-p2p #26: failed to connect to /192.168.49.1 (port 8988) from /192.168.49.1 (port 47274) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)

J-Ghodasara, 2020-07-02:
I am getting this error while sending the file or message.  I have followed exactly the same steps for sending a file as mentioned but still, I am not able to send it. Can you please help me out?

My code is as below. I was able to connect the two devices successfully but while sending file I am getting the above error.

```javascript
 sendMessage() {
    getConnectionInfo().then((info) => {
      console.log('Connection Info Send', info);
      if (info.groupOwnerAddress !== null) {
        ImagePicker.openPicker({
          cropping: false,
          includeBase64: true,
          mediaType: 'photo',
        })
          .then((image) => {
            console.log('Image', image);
            sendFile(image.path);
          })
          .catch((e) => {
            console.log('Photo picker error', e);
          });
      }
    });
  }
```

```javascript
onStartInvestigate = () => {
    PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
      {
        title: 'Access to wi-fi P2P mode',
        message: 'ACCESS_COARSE_LOCATION',
      },
    )
      .then((granted) => {
        if (granted === PermissionsAndroid.RESULTS.GRANTED) {
          console.log('You can use the p2p mode');
        } else {
          console.log('Permission denied: p2p mode will not work');
        }
      })
      .then(() => {
        startDiscoveringPeers()
          .then((val) => {})
          .catch((e) => {
            console.log('ERROR DISCOVERING PEERS', e);
          });
      })
      .then(() => {
        console.log(`Peers searching started`);
        subscribeOnPeersUpdates(({devices}) => {
          console.log(`New devices available: ${JSON.stringify(devices)}`);
          if (devices.length > 0) {
            connect(devices[0].deviceAddress)
              .then(() => {
                console.log('Successfully connected');
                getConnectionInfo().then((info) => {
                  console.log('Connection Info', info);
                  receiveFile().then((path) =>
                    console.log(`Path to file: ${path}`),
                  );
                });
              })
              .catch((e) => {
                console.log('Error connecting', e);
              });
          }
        });
        // getAvailablePeers().then((peers) => {
        //   console.log(peers);
        //   if (peers.devices.length > 0) {

        //   }
        // });
      })
      .catch((err) =>
        console.error(
          `Something is gone wrong. Maybe your WiFi is disabled? Error details: ${err}`,
        ),
      );
  };

```

kirillzyusko, 2020-07-02:
Hi @J-Ghodasara 
How `path` variable looks like?

J-Ghodasara, 2020-07-02:
The path looks like "file:///file%3A/data/user/0/com.juvo.sharesafe/cache/react-native-image-crop-picker/IMG-20200701-WA0003.jpg"

kirillzyusko, 2020-07-02:
Could you try to replace `%3A` symbol to corresponding UTF-8? As far as I remember it should be `:` (however I'm not 100% sure :))

J-Ghodasara, 2020-07-02:
I have tried this path "file:///data/user/0/com.juvo.sharesafe/cache/react-native-image-crop-picker/IMG-20200701-WA0003.jpg" but still getting same error.

kirillzyusko, 2020-07-02:
Ok, can you then capture native logs (via `adb logcat`) and send it here as file?

J-Ghodasara, 2020-07-02:
Yes sure. Here is the log file you can find the error in the end.
[logcat.docx](https://github.com/kirillzyusko/react-native-wifi-p2p/files/4862729/logcat.docx)

J-Ghodasara, 2020-07-02:
These are the steps I have done:-

1) Create a Group on 1st device.
2) Connect() on 2nd device.
3) ReceiveFile() called on 2nd device
4) SendFile() called on 1st device.

Correct me if I am doing anything different

kirillzyusko, 2020-07-02:
Hm, I found it in my docs:

```

1. Create group on server side (side, which will receive message) (1st device according your message)
2. Connect to server-group from client side (2nd device in your case)
3. After establishing connection you should call getConnectionInfo() on client side (also recommend do it on group-server side) (I hope you did it on both devices😊 )
4. Call receiveMessage() on server side (receiveMessage return Promise<string> - message from client) (you wrote, that you call it on 2nd device, but I wrote about "server" device - it should be 1st device in your case. Could you try to call receiveFile on your first device?)
5. Call sendMessage(message) on client side, after resolving getConnectionInfo() (it should be called on your 2nd device)

```

Could you change devices in 3-4 items and write here about results?

J-Ghodasara, 2020-07-02:
Also, I have noticed that I am also getting this error "Error while File sending {"code": 1, "message": "Operation failed because p2p is unsupported on the device."}".

But I have checked on both devices and both are having wifi direct options. So I am not able to understand that why it's saying that p2p is unsupported on device. Are you having idea why is it so?

kirillzyusko, 2020-07-02:
> Also, I have noticed that I am also getting this error "Error while File sending {"code": 1, "message": "Operation failed because p2p is unsupported on the device."}".

Is it happening when you call `sendFile`?

J-Ghodasara, 2020-07-02:
Yes when I am getting this when I call sendFile

kirillzyusko, 2020-07-02:
Hm, interesting 🤔 
Did you get the same error when you collecting `logcat` logs?
If yes, then a little bit later I promise to have a look on this problem (when I get to a laptop with windows😄 )

J-Ghodasara, 2020-07-02:
Yes, the error is present in the logcat file and it is at the end of the file so you can find it easily. No worries please have a look at it and till then I will try to debug the error and try to understand what's the problem. 

Please revert back to me if you find the solution because since the past 2 days this error has blocked me.
Thanks for your time :)

kirillzyusko, 2020-07-02:
@J-Ghodasara I can not open this file on Windows too.
Could you send these logs in `.txt` format?

J-Ghodasara, 2020-07-06:
@kirillzyusko I was trying to debug the issue and I found that inside FileServerAsyncTask.java line number 48 that is f.createNewFile(); at this line, I am facing some issue. After this line the code is not getting executed can you please check and tell what may be the issue.

The path I am sending in sendFile() is "/storage/emulated/0/DCIM/Screenshots/Screenshot_2020-07-06-10-33-29-28_3a637037d35f95c5dbcdcc75e697ce91.png".

kirillzyusko, 2020-07-06:
Hi @J-Ghodasara 
Can you request READ/WRITE permissions?

https://github.com/kirillzyusko/react-native-wifi-p2p#sendfilepathtofile

J-Ghodasara, 2020-07-06:
voilà man. Now it works. Awesome man thank you so much. Finally after 4 days of debugging its finally working i am so happy. Closing this issue.

closed 2020-07-06 by J-Ghodasara