kirillzyusko/react-native-wifi-p2p #27: App Crashing when clicking on Connect Button

sharmakumaraditya, 2020-07-02:
Whenever i am clicking the Connect , Create Group or Send Button the App is Crashing and when i am pressing other buttons nothing is happening, can anyone please help me with this. The app is crashing again and again and to no success i was able to fix it.

kirillzyusko, 2020-07-02:
Hi @sharmakumaraditya 
1. Does your device support p2p feature? Are you sure, that wifi enabled, when you call `createGroup`?
2. Did you call `init` method in the beginning?
3. Could you capture logs via `adb logcat` and send them here as `*.txt` file?

sharmakumaraditya, 2020-07-02:
link to the txt file : [log.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/4865876/log.txt)
1. Yes my Device does support p2p

Thanks for the quick reply

kirillzyusko, 2020-07-02:
It looks like `state.devices` is empty. Then `state.devices[0].deviceAddress` will give you an error.
Am I correct?

sharmakumaraditya, 2020-07-02:
I actually haven't checked that , i am just using this example https://github.com/kirillzyusko/react-native-wifi-p2p-example , i have just cloned this repo and actually i am trying to run this app.

kirillzyusko, 2020-07-02:
Can you attach log when you do following:
1. Open app
2. Call `init`
3. Call `createGroup`

sharmakumaraditya, 2020-07-02:
[log.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/4866419/log.txt)

This is the log file for create group

kirillzyusko, 2020-07-02:
@sharmakumaraditya cool. But I see only JS logs. I'd like to see all logs (including Java). Could you please capture it via `adb logcat` comand?

sharmakumaraditya, 2020-07-02:
Sure i got it, the app crashes when the create group is clicked more than once.

closed 2020-07-02 by sharmakumaraditya