kirillzyusko/react-native-wifi-p2p #28: File path and file name on receiver end

sandipchandra17, 2020-07-05:
Can you please let me know where can I change the receive file name and path as I see there is hardcoded path and extension is aleays jpg where the file is getting saved.

sandipchandra17, 2020-07-05:
And also why jpg file not reflected in gallery on receiver phine..

kirillzyusko, 2020-07-06:
Hi @sandipchandra17 
It's fixed in https://github.com/kirillzyusko/react-native-wifi-p2p/commit/ed7fe609ecfad62835dc545d0734c761ad5ae9af
I'm going to re-write a lot of code from the library in the nearest days.
So, I hope, soon I will publish version `3.0.0`

sandipchandra17, 2020-07-06:
> Hi @sandipchandra17
> It's fixed in [ed7fe60](https://github.com/kirillzyusko/react-native-wifi-p2p/commit/ed7fe609ecfad62835dc545d0734c761ad5ae9af)
> I'm going to re-write a lot of code from the library in the nearest days.
> So, I hope, soon I will publish version `3.0.0`

sandipchandra17, 2020-07-06:
Hi,

sandipchandra17, 2020-07-06:
I saw ur code in java files and I also tried to capture the filename from socket object in doinbackground method but not working.. are you to send hardcoded filename from app.js I see you used filename and path from receivefile method.. I just wanted to pick the extension whichever file selected by sender in the doinbackground mthod of fileserverasynctask.java file.

kirillzyusko, 2020-07-06:
The receiveFile method will take two parameters (the directory where file should be saved and the file name).
If you need to transfer the same file name that the client sends, you can use sendMessage/receiveMessage methods, where before sending the file, you should first send its name, which is then specified in the receiveFile.

I did this because native tools do not involve transferring a file with the same name, and you cannot do this with native tools. You can write wrappers around the main methods and encapsulate the logic with the transfer of the file name and other meta-information in the methods themselves. However, this approach contradicts the concept of this module. The purpose of this wrapper was to make the most minimal api that interacts with the main methods. If anyone needs it, then I don’t think it will be a problem to write my own flow to send the file.

Do not worries, I almost finished rewriting all pieces that I wanted to rewrite. I just need to recollect what I did, prepare changelog, migrations docs, change readme (including new code samples), publish this package, update example project.
I hope it will be done within next few hours.

kirillzyusko, 2020-07-06:
Fixed in `3.0.0`

closed 2020-07-06 by kirillzyusko

sandipchandra17, 2020-07-06:
Thanks sir..

I have one question how do I know which file extension sender has send like it can be mp3 or jpg or mp4.. so if I run this application in two mobile how the receiver mobile identify as I cant hardcode file extension..

kirillzyusko, 2020-07-06:
Call `sendMessage` before with your `fileName`.

So the flow is next: determine `fileName` on client side -> call receiveMessage() on server -> sendMessage(fileName) on client -> call receiveFile(directory, receivedFileName) on server -> call sendFile() on client.

sandipchandra17, 2020-07-06:
Thanks for your suggestion.  I tested it successfully as per your steps and different types of file are receiving..

But the problem is jpg or mp4 filesreceived though it is not reflected in mobile gallery automatically what generally happens but if i change the file name or the received folder name then its  reflected in gallery.. 
So you have any idea why it is happening in android phone and my android version is 9

kirillzyusko, 2020-07-06:
Hm, to be honest don't have any ideas :)
But I think it's not a problem of this library
I would say it's a problem of Android
Similar issue: https://stackoverflow.com/questions/36012618/android-renamed-image-file-not-shown-at-phone-gallery-folder
So I'd recommend you to search this problem in internet😉

sandipchandra17, 2020-07-07:
I saw it internet i think something has to add in the intent level: File file = ..... // Save file context.sendBroadcast(new Intent(Intent.ACTION_MEDIA_SCANNER_SCAN_FILE, Uri.fromFile(file))).

kirillzyusko, 2020-07-07:
@sandipchandra17 Can you share a link, where did you see it?
And, I think, it's better to open new issue, for the problem with appearance file in gallery.