kirillzyusko/react-native-wifi-p2p #30: Multiple file transfer at a time

sandipchandra17, 2020-07-07:
Is it possible to send mutiple files to receiver using this library?
Scenario is receiver will click one time in the receive file button and sender select multiple file and send at one go..

kirillzyusko, 2020-07-07:
Hm, I never thought about it.
But once again, you can use the trick with sending message, which contains all required information.
For example you can create own methods sendMultipleFiles(files[])/receiveMultipleFiles()

Let's consider next flow (how these functions may be implemented):
1. Server calls `receiveMessage()`
2. Client calls `sendMessage(JSON.stringify({ filesCount: files.length }))`
3. After receiving the message server trying to parse the message: `const { filesCount } = JSON.parse(message)`;
4. Then server execute next cycle: `for (let i = 0; i < filesCount; i++) { await receiveFile() }`
5. And client also in cycle send all files: `for (let i = 0; i < files.length; i++) { await sendFile(files[i]) }`

sandipchandra17, 2020-07-07:
In which file need to create these methods.

kirillzyusko, 2020-07-07:
You can create it in your project :)
It shouldn't be included in this library I think.
If it's needed in your project - create own set of functions.
Because every project is unique. And requirements for every project is also unique.
The purpose of this library is providing set of functions, which are written on the top of native API.
I wouldn't like to cover all cases in one library, since API will contain a lot of methods and it may be hard to support.
I hope I answered on your question☺️

sandipchandra17, 2020-07-07:
Thanks I will try in my react native code..

kirillzyusko, 2020-07-07:
Thanks. I will close this issue.
Feel free to re-open, if needed.

closed 2020-07-07 by kirillzyusko