kirillzyusko/react-native-wifi-p2p #31: Appearance files in receiver mobile gallery

sandipchandra17, 2020-07-07:
To display received file likes image or video in gallery we can look for the below:
Link: https://stackoverflow.com/questions/2170214/image-saved-to-sdcard-doesnt-appear-in-androids-gallery-app

kirillzyusko, 2020-07-07:
I got it
I think I will change the source of this library in the nearest days and publish it under `3.1.0` version (if I can not find another way for force-scanning gallery).

kirillzyusko, 2020-07-09:
Changes are shipped in https://github.com/kirillzyusko/react-native-wifi-p2p/commit/a89d517247055e6e75beefac29ecdf4fa9541a10
I'm going to test it today evening. If everything is fine I will publish it today evening.

sandipchandra17, 2020-07-09:
Great job.. thanks

kirillzyusko, 2020-07-09:
@sandipchandra17 If you want - you can test it and give me your feedback. Merely install this package from github and rebuild your app. And you will be able to use new feature with `forceToScanGallery`.
Also in this case you can save my time and I can spend it on fixing other issues that you reported :)

sandipchandra17, 2020-07-09:
I can't download the files. If you can provide me somehow or directly publish I will test and let you know.

kirillzyusko, 2020-07-09:
@sandipchandra17 sure, I can.
Just change this line in package.json:
```
"react-native-wifi-p2p": "github:kirikzyusko1996/react-native-wifi-p2p#master"
```
And re-install your dependencies.

sandipchandra17, 2020-07-09:
Installed successfully 3.1.0 but getting an error when i click on receive file. I am passing true as 3rd argument from app.js receive file method. 
[image]

I think this is due to one callback you added in manager module- receiveFile method but index.js sending 3 argument.

kirillzyusko, 2020-07-09:
index.js sending 4:     

`WiFiP2PManager.receiveFile(folder, fileName, forceToScanGallery, (pathToFile) => {`

Can you send me code snippet how do you use new method?

sandipchandra17, 2020-07-09:
Sending true from app.js - receiveFile method.
[image]

sandipchandra17, 2020-07-09:
Passing true from receiveFile method in app.js file:

[image]

kirillzyusko, 2020-07-09:
Weird.
I will test it.

sandipchandra17, 2020-07-09:
Ok.. also along with this you can check this in yiur test if the previous one not working.

I found another info for different androis version not sure helpful or not:::
public void refreshAndroidGallery(Uri fileUri) {
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.KITKAT) {
            Intent mediaScanIntent = new Intent(
                    Intent.ACTION_MEDIA_SCANNER_SCAN_FILE);
            mediaScanIntent.setData(fileUri);
            mContext.sendBroadcast(mediaScanIntent);
        } else {
            mContext.sendBroadcast(new Intent(
                    Intent.ACTION_MEDIA_MOUNTED,
                    Uri.parse("file://" + Environment.getExternalStorageDirectory())));
        }
    }

Link: https://stackoverflow.com/questions/4144840/how-can-i-refresh-the-gallery-after-i-inserted-an-image-in-android

kirillzyusko, 2020-07-10:
Will be shipped in 3.1.0

closed 2020-07-10 by kirillzyusko

sandipchandra17, 2020-07-10:
I tested this in android 7 and android 9 mobile working fine.. only one mobile lenovo k5note android 5 not refreshing gallery dont know why. May be mobile issue..

kirillzyusko, 2020-07-10:
Or android specific version issue. I don't have a lot of android devices. Only Xiaomi and Huawei.
On both everything was working fine.
That's why I published new version.

sandipchandra17, 2020-07-10:
Okay fine then even my 2 android phones working fine in sumsung and geonee.