kirillzyusko/react-native-wifi-p2p #32: Message and file not working all time

sandipchandra17, 2020-07-07:
1. I followed the steps receive msg->send msg but getting differebt errror in different times though my connection is well formed between two mobile.
I attached screenshots
[image]

[image]

2nd issue: file not received sometimes and damagrd file or half file received
[image]
[image]

kirillzyusko, 2020-07-07:
Try to add delay before sending file. I think ~300ms is enough.

sandipchandra17, 2020-07-07:
300ms is not working. 
Also after 3secs i am geeting error as below
[image]

Sometimes long waiting its working but no guarntee sometimes getting this error .

kirillzyusko, 2020-07-07:
Can you share your code?

sandipchandra17, 2020-07-08:
Dear Kirill,

I have sent my app.js via gmail to your gmail.. please check.

kirillzyusko, 2020-07-08:
@sandipchandra17 You sent me almost identical code from my example project. Which flow isn't working there?

sandipchandra17, 2020-07-08:
As i told different types of error like below scenario:
1. Testing in two mobile android 9 and another android 6.
Clicked receive msg in receiver mobile -> clicking on send messege on sender mobile, its saying files successfully received in receiver mobile but nothing received.
2. Sometimes clicking on send msg and receiver mibile received exif. Screenshot attached.
3. Send msd>receive msg sucessful and receive file fine but after that when clicking send file msg successfully received but file received damaged or half file.
4. If first time connect issue occurs then nothing will happed msg or file send.. i have to again remove group then problem solved but have to wait for long time but no gurantee above points occuring almost 80% case.

kirillzyusko, 2020-07-08:
1. I think it's impossible. "successful" means no error occurred in Java code. So file should be in a place.
2. It seems, like stream of bytes was interrupted or you are trying to send the message not in UTF8. I need to have a look on Java logs. I can not say anything without them.
3. That would be nice if you could re-create this scenario in code. By promise chaining. In this case I can reproduce it. During my testing everything worked fine.
4. If you can not connect, then it's no sense to try to send messages/files. Or I misunderstood the description.

To sum it up: that would be nice, if you could share real example of code. Right now it sound too abstract and I can not even reproduce these issues. As a result I can nit help you nevertheless I want...
So sharing code snippet would be really nice.

sandipchandra17, 2020-07-08:
This is the same code i sent younin gmail.
1. In this scenario why received file message would come when i click on send messge. It should come as message received successfully.
2. I am sending normal string message if you can look my code send message method.
3. Sometimes its working fine sometimes not may be time gap issue.. but in real life application how do i know what time i have to put.. 
4. Agreed. I will check again using debug mode..

kirillzyusko, 2020-07-08:
The code that you sent me presented in format button -> handler. You haven't provided a full history of all actions that you performed/buttons that you pressed. I am asking for small piece of code. Can you create your own button (like "Reproduce Issue 1") and make handler for it, where I will see all methods, which you are calling and the order of these methods?

sandipchandra17, 2020-07-08:
Sorry I dont get your point.. what do you mean by firmat button-handler. I am testing same file you provided in the site..
 My scenario is I have generated apk file from that code and installed in two mobiles. Then the process you mentioned I am doing

kirillzyusko, 2020-07-08:
When I tested everything was fine, even with multiple message sending.
At least you are trying to send messages two times, right?
Can you put this business logic into separated function, and create special button for calling this method?
And provide this code here?

sandipchandra17, 2020-07-08:
I havnt started any logic on yiur cide yet.
What I did is single click on below button clicks on the same code app.js yiu provided.. I dint how everything was fine in your test but i am testing same code.

My steps only one click:
Create group in mobile 1
Investigate in mobile 2
Connect in mobile 2
Receive message in mobile 1
Send message in mobile 1 --- giving above errors exif and sometimes no msg received..in apk app is closing.
Receive file
Send file sometimes ok sometimes corrupted file received sometimes nothing happens.

kirillzyusko, 2020-07-08:
Hm, did you call `getConnectionInfo` after connection established?

sandipchandra17, 2020-07-09:
Yes I checked connection info and it was fine.. 
Today I tested again with some more time gap between steps and its working fine.. but how come end user will know how much have to wait..
So what I qas thinking Is it possible from your java classes return some messages like when receive message arrived successfully in receiver mobile, sender will be notified so that app will allow him/her to proceed for file upload.. or if one file is still tranferring sender and receiver can get transfer status so that  app can restrict to do further action until its finish operation through react native..

And another point can you please check in java classes as  i dont know much java why sometimes send message giving file received successfully. Is it same method used for both file and message.. I found this when the scenaio like if a file received half like damaged file then if i click receve msg->send message it is occuring..

kirillzyusko, 2020-07-09:
Hi @sandipchandra17 
In fact your message gave me one interesting idea, why it may happen.
If I try to fix one bug and give you a branch (I won't publish these changes in npm but you will be able to install it from github) - can you test it and say, whether bug is persist?

sandipchandra17, 2020-07-09:
Ok I will test this more minutely.. will let you know..

sandipchandra17, 2020-07-09:
I was thinking the point that rather than increasing time can we have live status exposed to both sender and received about message send to receiver or not informed to sender and file is downloading status to both sender and receiver and how much data is remaining. I don't know its possible or not just I thought.

kirillzyusko, 2020-07-09:
@sandipchandra17 It can be possible. I can emit event when part of data has been transferred successfully.
But promises model is informing us about that. When promise is resolved -> data transferring is ended.

To be honest I saw the time difference between 'File received successfully' and 'File sent successfully:' ~300ms, when I transferred file ~10mb. I can not understand the cause of this delay, but I have several assumptions why it may happen. I see next:
- async RN bridges;
- network delays.

Apart of this I noticed, that sockets are closing after promise resolving. I think it's also incorrect and it may be a cause of damaged file. And I want to fix the socket closing. But I can not test my changes.
That's why I am asking you to help me with it.

sandipchandra17, 2020-07-10:
Yes I will test it.. let me know once you are done.

kirillzyusko, 2020-07-10:
Discovered one problem. I'll try to fix it.

sandipchandra17, 2020-07-10:
If possible please emit transfered byte so that it can be shown in progressbar to sender and receiver using react native..

kirillzyusko, 2020-07-11:
Small update: my laptop has died😔
I'm going to send it to the service center within next few days (two days I hope).
Hope they can fix it quickly.

sandipchandra17, 2020-07-12:
Oh sad.. wish it recover soon..

sandipchandra17, 2020-07-12:
Hi Kirill,

Need one clarification:
Is this library support that sender can connect with multiple receiver and send same files to all receivers at the same time. 
Or this feature not currently included?

kirillzyusko, 2020-07-12:
Need to test it :)

sandipchandra17, 2020-07-13:
Another two questions and one observation:
1. Is it possible to get own device name.
2. Is it possible for receiver know the device name of which sender connected. Because from group info its not clear. 
3. Observation: when receiver successfully remove group but after long time later group info updated to sender means null.
Is it normal as wifi direct taking time for this update or bug?

kirillzyusko, 2020-07-13:
1. After enabling WiFI you should receive `THIS_DEVICE_CHANGED_ACTION`. According to this answer there should be info that you need.

For 2-3 I will have a look later.

P. S. don't mix your issues into one place. Feel free to create different issues :)

sandipchandra17, 2020-07-13:
Created 3 issues #34, #35 and #36

Rk-choudhary-ck, 2020-10-10:
Hii 
i am facing the same problem in my project. when i click on send message i will face this problem. please tell me what should i do? and one thing more i don't know how i test my code because i'm running my project on real device. it is connect with multiple wifi connections but message not send . just i used the code from github with no changes . can i need to change any part of my code so my message or file send successfully.
please help me.

kirillzyusko, 2020-10-10:
@Rk-choudhary-ck The problem above was happening because `sendMessage` method was calling several times. Could you describe your steps more precisely? How many devices do you have? What is the error do you get? Which steps are you doing?

Rk-choudhary-ck, 2020-10-12:
i am using the two android device my connection is established successfully but i am not share any message or file one to another device . could you tell me how should i start from first step.

Rk-choudhary-ck, 2020-10-12:
[image]

kirillzyusko, 2020-10-12:
Hi @Rk-choudhary-ck 
Did you follow [this](https://github.com/kirillzyusko/react-native-wifi-p2p#sendmessagemessage) instruction?
Are you sure, that you completed 3d item in this list?

Rk-choudhary-ck, 2020-10-12:
Hii Kirillzyusko
yes i following these instructions but i get a error about fail connection the error is :
 Error while message sending [Error: failed to connect to /192.168.49.1 (port 8988) from /192.168.49.1 (port 38749) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)]

kirillzyusko, 2020-10-12:
> [Error: failed to connect to /192.168.49.1 (port 8988) from /192.168.49.1 (port 38749) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)]

@Rk-choudhary-ck When this error happens?
Could you send adb logcat logs as well?

Rk-choudhary-ck, 2020-10-12:
i think it's no about logcat because it will happen in case when i send message .

kirillzyusko, 2020-10-12:
I was asking for logs from logcat in order to understand where is the error happens...
Without logs from device I can not say anything, unfortunately.

Rk-choudhary-ck, 2020-10-13:
Hii kirillzyusko
now i am face in log this kind of error.
 LOG      Error while file sending [Error: failed to connect to /192.168.49.1 (port 8988) from /192.168.49.1 (port 35053) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)]

kirillzyusko, 2020-10-13:
@Rk-choudhary-ck I can not help unless you provide `adb logcat` output. You give me only final exception. But I can not understand what is the reason of this exception. Maybe it's permission issues. Maybe something another. I don't know. That's why I ask you to give me logs from your device.

Rk-choudhary-ck, 2020-10-13:
Thanks Kirillzyusko

Rk-choudhary-ck, 2020-10-22:
Hello Sir..
Now p2p working well for me. i wanna make a aap like shareit .. onSendfile
method not send file when i use document picker package .. i think file
path may be incorrect . so please give me any solution what can i do for
share all types of files.

kirillzyusko, 2020-10-22:
Hi @Rk-choudhary-ck 
Can you please give an example of path that doesn't work?

Rk-choudhary-ck, 2020-10-22:
[{"fileCopyUri":
"content://com.android.providers.media.documents/document/audio%3A34349",
"name": "Swaah Ban Ke (MzcPunjab.Com) AETrim1554384038160", "size": 801597,
"type": "audio/mpeg", "uri":
"content://com.android.providers.media.documents/document/audio%3A34349"}]

this is complete information about file path that stored in multipleFile
variable ... but i'm only using this path
content://com.android.providers.media.documents/document/audio%3A34349

i also use this path append type of this file at end of the path. but it's
not work for me .

kirillzyusko, 2020-10-22:
Hi @Rk-choudhary-ck 
Please, have a look on [this](https://github.com/kirillzyusko/react-native-wifi-p2p/issues/38#issuecomment-678618178) thread
Your path contains symbols like `%3A`. You should convert them to plain chars.

Rk-choudhary-ck, 2020-10-22:
sir one thing more i set path of storage in onReceiveFile function
return receiveFile(RNFS.DownloadDirectoryPath, 'BFMV:Letting You Go.mp3')

can this storage receive all types of files that i have selected ?

Rk-choudhary-ck, 2020-10-22:
sir now i make exact format of file but still have a problem my log is show
:-

content://com.android.providers.media.documents/document/audio:3A34349
[Thu Oct 22 2020 16:37:15.223]  LOG      You can use the storage
[Thu Oct 22 2020 16:37:15.437]  LOG      File received successfully

but file not show in the folder .. that means file has not been sent

kirillzyusko, 2020-10-22:
What is the path `RNFS.DownloadDirectoryPath` looks like?

Rk-choudhary-ck, 2020-10-23:
this one
/storage/emulated/0/Download

kirillzyusko, 2020-10-23:
@Rk-choudhary-ck you missed a slash. It should be `/storage/emulated/0/Download/`.

Rk-choudhary-ck, 2020-10-23:
Hello sir ..
i facing now same error after changes onSendFile and onReceiveFile these
are my paths:-
file Receive path: /storage/emulated/0/Download/
file SendPath:
content://com.android.providers.media.documents/document/audio:3A34349
and message is displayed: File received successfully
but file not send actually..

kirillzyusko, 2020-10-23:
@Rk-choudhary-ck if you want my help, then, please, attach `adb logcat` logs for both devices. For sender and for receiver.

Rk-choudhary-ck, 2020-10-24:
sir these are my both files logcat and console file .. please check these
file if any mistake please tell me.

[Sat Oct 24 2020 10:28:27.671]  BUNDLE  ./index.js

[Sat Oct 24 2020 10:28:36.297]  BUNDLE  ./index.js

[Sat Oct 24 2020 10:28:39.175]  LOG      Running "shareit" with {"rootTag":1}
[Sat Oct 24 2020 10:28:39.179]  LOG      You can use the p2p mode
[Sat Oct 24 2020 10:28:39.382]  LOG      startDiscoveringPeers status:  success
[Sat Oct 24 2020 10:28:39.802]  LOG      Running "shareit" with {"rootTag":1}
[Sat Oct 24 2020 10:28:42.376]  LOG      You can use the p2p mode
[Sat Oct 24 2020 10:28:42.485]  LOG      startDiscoveringPeers status:  success
[Sat Oct 24 2020 10:28:46.124]  LOG      OnPeersUpdated [{"deviceAddress": "72:3a:51:fe:1e:15", "deviceName": "Redmi", "isGroupOwner": false, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}]
[Sat Oct 24 2020 10:28:58.724]  LOG      OnPeersUpdated [{"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "Android_c4c1", "isGroupOwner": false, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}]
[Sat Oct 24 2020 10:29:10.335]  LOG      Group created successfully!
[Sat Oct 24 2020 10:29:10.611]  LOG      THIS_DEVICE_CHANGED_ACTION {"interface": "p2p0", "networkName": "DIRECT-Um-Android_c4c1", "owner": {"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "", "primaryDeviceType": null, "secondaryDeviceType": null, "status": 4}, "passphrase": "stPb0buN"}
[Sat Oct 24 2020 10:29:10.620]  LOG      OnConnectionInfoUpdated {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": true}
[Sat Oct 24 2020 10:29:16.596]  LOG      [{"fileCopyUri": "content://com.android.providers.media.documents/document/audio%3A34349", "name": "Swaah Ban Ke (MzcPunjab.Com) AETrim1554384038160", "size": 801597, "type": "audio/mpeg", "uri": "content://com.android.providers.media.documents/document/audio%3A34349"}]
[Sat Oct 24 2020 10:29:18.227]  LOG      {"devices": [{"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "Android_c4c1", "isGroupOwner": false, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}]}
[Sat Oct 24 2020 10:29:19.626]  LOG      Connect to:  {"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "Android_c4c1", "isGroupOwner": false, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}
[Sat Oct 24 2020 10:29:19.654]  LOG      Successfully connected
[Sat Oct 24 2020 10:29:19.678]  LOG      OnPeersUpdated [{"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "Android_c4c1", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 1}]
[Sat Oct 24 2020 10:29:20.680]  LOG      OnPeersUpdated [{"deviceAddress": "72:3a:51:fe:1e:15", "deviceName": "Redmi", "isGroupOwner": false, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 0}]
[Sat Oct 24 2020 10:29:20.950]  LOG      OnConnectionInfoUpdated {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": true}
[Sat Oct 24 2020 10:29:20.187]  LOG      OnPeersUpdated [{"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "Android_c4c1", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 0}]
[Sat Oct 24 2020 10:29:20.232]  LOG      THIS_DEVICE_CHANGED_ACTION {"interface": "p2p0", "networkName": "DIRECT-Um-Android_c4c1", "owner": {"deviceAddress": "da:32:e3:eb:16:20", "deviceName": "Android_c4c1", "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 4}, "passphrase": ""}
[Sat Oct 24 2020 10:29:20.759]  LOG      OnConnectionInfoUpdated {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": false}
[Sat Oct 24 2020 10:29:23.130]  LOG      getConnectionInfo {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": false}
[Sat Oct 24 2020 10:29:28.245]  LOG      You can use the storage
[Sat Oct 24 2020 10:29:28.357]  LOG      mypath: /storage/emulated/0/Download/
[Sat Oct 24 2020 10:29:30.373]  LOG      content://com.android.providers.media.documents/document/audio:3A34349
[Sat Oct 24 2020 10:29:30.386]  LOG      You can use the storage
[Sat Oct 24 2020 10:29:30.712]  LOG      File received successfully
--------- beginning of crash
--------- beginning of system
--------- beginning of main
10-24 10:28:35.368 19715 19983 I ReactNativeJS: Running "shareit" with {"rootTag":1}
10-24 10:28:35.874 19715 19715 I ReactNative: [GESTURE HANDLER] Initialize gesture handler for root view com.facebook.react.ReactRootView{8d272ee V.E...... ......ID 0,0-720,1296 #1}
10-24 10:28:35.883 19715 19983 I ReactNativeJS: You can use the p2p mode
10-24 10:28:36.115 19715 19983 I ReactNativeJS: 'startDiscoveringPeers status: ', 'success'
10-24 10:28:55.464 19715 19983 I ReactNativeJS: 'OnPeersUpdated', [ { status: 3,
10-24 10:28:55.464 19715 19983 I ReactNativeJS:     secondaryDeviceType: null,
10-24 10:28:55.464 19715 19983 I ReactNativeJS:     deviceAddress: 'da:32:e3:eb:16:20',
10-24 10:28:55.464 19715 19983 I ReactNativeJS:     isGroupOwner: false,
10-24 10:28:55.464 19715 19983 I ReactNativeJS:     primaryDeviceType: '10-0050F204-5',
10-24 10:28:55.464 19715 19983 I ReactNativeJS:     deviceName: 'Android_c4c1' } ]
[... cut ...]
10-24 10:29:17.492 19715 19983 I ReactNativeJS:   groupOwnerAddress: { isLoopbackAddress: false, hostAddress: '192.168.49.1' } }
10-24 10:29:19.825 19715 19983 I ReactNativeJS: 'getConnectionInfo', { isGroupOwner: false,
10-24 10:29:19.825 19715 19983 I ReactNativeJS:   groupFormed: true,
10-24 10:29:19.825 19715 19983 I ReactNativeJS:   groupOwnerAddress: { isLoopbackAddress: false, hostAddress: '192.168.49.1' } }
10-24 10:29:27.101 19715 19983 I ReactNativeJS: content://com.android.providers.media.documents/document/audio:3A34349
10-24 10:29:27.114 19715 19983 I ReactNativeJS: You can use the storage

Rk-choudhary-ck, 2020-10-26:
Hello sir
 i am waiting for your reply

kirillzyusko, 2020-10-26:
@Rk-choudhary-ck well, first of all - you attached JS logs/logs from metro bundler. It's not sufficient information and I can not tell anything from this logs. Secondly - your problem is that you can not properly use two libraries together (react-native-document-picker and react-native-wifi-p2p) and it's not related to this issue (the problem in this issue is that sending multiple files with short delay cause broken files).
So, **PLEASE**, create new issue and attach `adb` logs. [Here](https://support.citrix.com/article/CTX232375) is a brilliant article how to do it, but I'm pretty sure you can find a lot of other.
I will ignore any new posts here that are not related to the problem described in the first post of this thread.
Thanks in advance.

Rk-choudhary-ck, 2020-10-26:
okay sir thanks
but please tell me what should  to do  for sharing multiple files p2p
without using react-native-document-picker. because i wanna make  app like
shareit and it's my first app.. please suggest any other package for choose
files and also work with react-native-wifi-p2p.
i'm thankful to you  because you giving me lot of time from your busy
schedule.

kirillzyusko, 2020-10-26:
@Rk-choudhary-ck please, create a new issue.

ChandanBeraTechno, 2022-07-08:
Hii kirillzyusko,
now i am face in log this kind of error.
 Error while message sending [Error: failed to connect to /192.168.49.1 (port 8988) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)]
 
 LOG  You can use the p2p mode
 LOG  startDiscoveringPeers status:  success
 LOG  Running "BobakWifiDDemo" with {"rootTag":1}
 LOG  You can use the p2p mode
 LOG  startDiscoveringPeers status:  success
 LOG  OnPeersUpdated [{"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "Redmi note 4", "isGroupOwner": false, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}, {"deviceAddress": "0c:54:15:e6:39:6d", "deviceName": "INDLAPTOP333", "isGroupOwner": true, "primaryDeviceType": "7-0050F200-0", "secondaryDeviceType": null, "status": 3}]
 LOG  OnPeersUpdated [{"deviceAddress": "0c:54:15:e6:39:6d", "deviceName": "INDLAPTOP333", "isGroupOwner": true, "primaryDeviceType": "7-0050F200-0", "secondaryDeviceType": null, "status": 3}, {"deviceAddress": "ae:c1:ee:63:51:69", "deviceName": "Redmi note 3", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}] 
 LOG  OnPeersUpdated [{"deviceAddress": "0c:54:15:e6:39:6d", "deviceName": "INDLAPTOP333", "isGroupOwner": true, "primaryDeviceType": "7-0050F200-0", "secondaryDeviceType": null, "status": 3}, {"deviceAddress": "ae:c1:ee:63:51:69", "deviceName": "Redmi note 3", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}] 
 LOG  Group created successfully!
 LOG  THIS_DEVICE_CHANGED_ACTION {"interface": "p2p0", "networkName": "DIRECT-Ht-Android_abfa", "owner": {"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "", "primaryDeviceType": null, "secondaryDeviceType": null, "status": 4}, "passphrase": "6eu23hhL"}
 LOG  OnConnectionInfoUpdated {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": true}
 LOG  OnPeersUpdated [{"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "Redmi note 4", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}, {"deviceAddress": "0c:54:15:e6:39:6d", "deviceName": "INDLAPTOP333", "isGroupOwner": true, "primaryDeviceType": "7-0050F200-0", "secondaryDeviceType": null, "status": 3}] 
 LOG  Connect to:  {"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "Redmi note 4", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 3}
 LOG  Successfully connected
 LOG  OnPeersUpdated [{"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "Redmi note 4", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 1}, {"deviceAddress": "0c:54:15:e6:39:6d", "deviceName": "INDLAPTOP333", "isGroupOwner": true, "primaryDeviceType": "7-0050F200-0", "secondaryDeviceType": null, "status": 3}] 
 LOG  OnPeersUpdated [{"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "Redmi note 4", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 0}, {"deviceAddress": "0c:54:15:e6:39:6d", "deviceName": "INDLAPTOP333", "isGroupOwner": true, "primaryDeviceType": "7-0050F200-0", "secondaryDeviceType": null, "status": 3}] 
 LOG  OnConnectionInfoUpdated {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": true}
 LOG  getConnectionInfo {"groupFormed": true, "groupOwnerAddress": {"hostAddress": "192.168.49.1", "isLoopbackAddress": false}, "isGroupOwner": true}
 LOG  OnPeersUpdated []
 LOG  OnPeersUpdated []
 LOG  Error while message sending [Error: failed to connect to /192.168.49.1 (port 8988) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)]
 LOG  OnPeersUpdated [{"deviceAddress": "e6:46:da:85:bc:eb", "deviceName": "Redmi note 4", "isGroupOwner": true, "primaryDeviceType": "10-0050F204-5", "secondaryDeviceType": null, "status": 0}]

kirillzyusko, 2022-07-10:
Seems like you can not establish a connection within 5s.
Why OnPeersUpdated reports empty array in the end?