kirillzyusko/react-native-wifi-p2p #33: CancelConnect() not working.

J-Ghodasara, 2020-07-10:
After successfully executing sendMessage() I am calling cancelConnect() to cancel the connection but it is throwing error - 

{"code":2,"message":"Operation failed because the framework is busy and unable to service the request"}.

Can you please help me out.

kirillzyusko, 2020-07-10:
Hi @J-Ghodasara 
It seems to be truth. I can reproduce it too.
I will check why it may happen and post my answer here.

kirillzyusko, 2020-07-11:
Small update: my laptop has died😔
I'm going to send it to the service center within next few days (two days I hope).
Hope they can fix it quickly.

J-Ghodasara, 2021-04-26:
Hello @kirillzyusko 

Did you get a chance to look at this issue?

kirillzyusko, 2021-04-26:
Hi @J-Ghodasara 

I already don't have second Android device, so I can not test it 😔