kirillzyusko/react-native-wifi-p2p #37: Cannot connect devices, plus getting an error in Android O

pratjosh9, 2020-07-15:
Cannot connect to groups.

In Android O, I'm getting an error "Operation failed due to an internal error"; after the location permission is provided.

Screenshot of the Error
[image]

**Note:** 
1. My repo containing the code can be found [here](https://github.com/pratjosh9/FileWarp)
2. The apk for the sample app can be downloaded from [here](https://github.com/pratjosh9/FileWarp/releases/tag/1.0.0)

How do I connect two devices running this app? I'm not able to join any groups.

kirillzyusko, 2020-07-15:
Hi @pratjosh9 
As I can see from screenshot error happens in: https://github.com/pratjosh9/FileWarp/blob/master/App.js#L81

But in fact you call a lot of methods (from line 61 till line 82), such as: `initialize`, `PermissionsAndroid.request`, `startDiscoveringPeers`. From the error description, I may assume it's thrown by `startDiscoveringPeers`. Am I correct?

I tested it on Android O too, and everything was working fine.

By some reasons native method is throwing an error, with code 0. You can read here about it: https://developer.android.com/reference/android/net/wifi/p2p/WifiP2pManager#ERROR
As you can see, there is no additional info, why it's failed.
Would be really nice, if you could attach logs from `logcat` tool. in this case I can say more about your problem.

BTW do you see nearby devices from settings?

pratjosh9, 2020-07-15:
Hi @kirillzyusko

Thanks for the quick response. 

I'll try to refactor the code to pinpoint the exact line resulting in Code:0 but most likely it is coming from `startDiscoveringPeers`.

Yes, I can see the wifi network created by  `createGroup` in settings.

[Here](https://paste.ee/p/nWgwu) is the logcat of some operations on the app. I tried `createGroup`, `removeGroup`, `getConnectionInfo`, `getGroupInfo` operations.

kirillzyusko, 2020-07-16:
@pratjosh9 Thanks for logs!

I have a question: do you have GPS enabled?
I found this in logs:
```
Location mode is disabled for the device
```
From certain Android versions it's required to have GPS enabled in order to be able to discover P2P devices (I wrote about this requirement in readme: https://github.com/kirillzyusko/react-native-wifi-p2p#caveats)

pratjosh9, 2020-07-16:
@kirillzyusko Thank you for the support.

I enabled location and changed parameters to few calls that led to the sample app being functional.

I'm closing this issue since the problem has been resolved.

closed 2020-07-16 by pratjosh9