kirillzyusko/react-native-wifi-p2p #38: Unable to connect [Error: failed to connect to /192.168.49.1 (port 8988) from /192.168.49.1 (port 37003) after 5000ms: isConnected failed: ECONNREFUSED (Connection refused)]

c-thun, 2020-08-14:
Repeatedly getting this while using sendMessage() and sendFile().
My code is as follows -
```
export const sendFilesToServer = (macAddr, files) => {
    console.log(files, macAddr);
    files.forEach(file => {
        WiFiP2P.connect(macAddr)
            // Resolve connection details using getConnectionInfo
            // .then(() => WiFiP2P.createGroup())
            .then(() => WiFiP2P.getConnectionInfo())
            .then(() => WiFiP2P.getAvailablePeers())
            // Send file metadata using sendMessage()
            .then(() => console.log(file.name))
            .then(() => {
                console.log('SENDING message');
                return WiFiP2P.sendMessage(file.name);
            })
            // Send file
            .then(() => {
                console.log('SENDING file');
                return WiFiP2P.sendFile(file.uri);
            })
            .catch(err => console.log('Unable to connect', err));
    });
}

export const receiveFromClient = () => {
    console.log('Can receive now');
    // Receive file metadata using receiveMessage()
    WiFiP2P.createGroup()
        .then(() => WiFiP2P.receiveMessage())
        // .then(() => {
        //     return PermissionsAndroid.request(
        //         PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE,
        //         {
        //             'title': 'Access to write',
        //             'message': 'WRITE_EXTERNAL_STORAGE'
        //         }
        //     )
        // })

        // Receive file and store it based on the metadata
        .then(message => {
            console.log('Received message ', message);
            return WiFiP2P.receiveFile('/storage/emulated/0/Music/', message);
        })
        .catch(err => console.log(err));
}
```

kirillzyusko, 2020-08-14:
Hm, looks like race condition in fact.
You are trying to send multiple files at the same time (that's how `forEach` works) - it's not correct.
In `receiveFromClient` function you receive only one file.

In case if you want to send multiple files you should do it sequentially: file1 -> file2 -> file3 (only after successful transfer of `file1` you start transfering `file2`).

I hope I gave you an answer on your question why it's not working. Try to rewrite your code and send files sequentially. Feel free to ask other questions if something is not clear :)

c-thun, 2020-08-14:
So if i try and send one file, it should work, right?

kirillzyusko, 2020-08-15:
> However, that still doesn't seem to explain why calling createGroup() is failing in the receiver side.

That would be nice if you can share your logs from `adb logcat`. Please, attach them as *.txt file.

You described your issue, and I understand what is the problem. And, of course, this problem shouldn't exist. So attaching Android logs (Java) can help me significantly :)

BTW, have you tried to switch you server/client devices? I mean run client code on "server" phone, and "client" code on server device? Merely change roles of devices. In this case you still receive the same error?

c-thun, 2020-08-15:
I did change the roles of the devices multiple times but it didn't do anything different except for once when it did actually manage to send a message to the other device. Other than that one transmission, it didn't work regardless of which device was client and which was server.

Will attach the ADB Logcat response in a few hours (away from my laptop right now)

c-thun, 2020-08-16:
Here is the ADB Logcat. Steps done in sequence : -
* Reloaded the app,
* Selected a file to send (the other device was on receive mode),
* Waiting for the other device to show up as peer,
* Selecting it and sending it (only one message and file to prevent race condition)
* App crashed while sending,
* Reloaded app, 
* Clicked on the receive button to initiate createGroup() which then failed.

In addition, for some reason it shows a FINE_LOCATION issue in adb logcat but I have provided both <uses-permission> and <uses-feature> for FINE_LOCATION in the Manifest and have provided the app with full location access (Allow all the time) after calling the location permission as asked from your code (And it shows that P2P mode can be used, isInitialised: true and startDiscoveringPeers successful
[output.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5081283/output.txt)

kirillzyusko, 2020-08-16:
As I wrote in https://github.com/kirillzyusko/react-native-wifi-p2p#sendmessagemessage you should call `getConnectionInfo` before sending actual data. Have you completed this step? As I see from logs the internal variable is `null`. That's why I thought you might have forgotten about this step.

c-thun, 2020-08-18:
```getAvailablePeers()``` was called only as a debugging step to confirm if it was actually seeing the other device or not. The logcat output after removing the line are attached below. Also, I used ```console.log(devices)``` in ```subscribeOnPeersUpdates()``` to confirm if there actually are devices nearby. Attaching that as well.

Also, you are right that ```getConnectionInfo()``` is failing
[adb_output.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5092623/adb_output.txt)
[image]

c-thun, 2020-08-18:
Trying after reversing the devices, here are the adb logcat files after reversing the devices roles. 

Also, I got a doubt. ```createGroup()``` always fails as it says that the framework is busy. Could it be because ```subscribeOnPeersUpdates()``` is always running in the background and preventing ```createGroup()``` from using the WiFiP2P service? If that is the case, I can unsubscribe before calling ```createGroup()``` and try

[reverse_adb_output.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5092642/reverse_adb_output.txt)

kirillzyusko, 2020-08-18:
> Also, I got a doubt. createGroup() always fails as it says that the framework is busy. Could it be because subscribeOnPeersUpdates() is always running in the background and preventing createGroup() [...]
I don't think so.
In fact I found the same problem. Try to find 01:49:44.409 from previous log. What was this command? createGroup? Have you checked - maybe group already was formed?
BTW, what type of debugging do you use? Through WiFi or through USB cable? (If you are debugging/downloading bundle via WiFi - please, try to switch to cable method).
I'm asking because I see `SoftException` from RN straight after the line that I mentioned (It's about WebSockets).

c-thun, 2020-08-18:
The one on 01:49:44.409 is the output that is caused by problems in the ```startDiscoveringPeers()``` method. This one actually would have happened because the app crashed during ```sendMessage()``` and upon reload, was not able to even start discovering the peers properly. I needed to force close the app and reload it from scratch to get it working normally. So, this output was post crash

Also, I am using USB cable (that came with the phone's charger) to debug the app. The soft exception would have been due to the crash I suppose

kirillzyusko, 2020-08-19:
One interesting moment:

```
Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["WIFI_P2P:PEERS_UPDATED",{"devices":[{"status":1,"isGroupOwner":false,"secondaryDeviceType":null,"primaryDeviceType":"10-0050F204-5","deviceAddress":"82:58:f8:25:e0:01","deviceName":"Android_819e"}]}])
```

Why bridge has been destroyed? :) It happens before actual sending.

Another moment:

```
08-19 01:49:33.109   674  1086 D WifiP2pService: Client:com.meridio is removed
08-19 01:49:33.110   674  1086 D WifiP2pService: Client:com.meridio is removed
```

Why it removes two times?

In adb reverse:

```
08-19 01:56:58.512 16570 18232 W ReactNativeJS: Cannot connect to the Metro server.
```
Why so?

I don't see any direct errors, to be honest... Could you edit your code in `node_modules` and add more `Log.i` statements? I don't know why `getConnectionInfo` failing...
Also I don't see `Successfully created group`.

I know, my answer probably will not help you, since I don't know what is going wrong...

c-thun, 2020-08-19:
So you mean adding more log statements in ``` react-native-wifi-p2p``` package?

And yes, I am never able to create group despite following the exact same order that was in your documentation. Can the issue here also be caused by the fact that I'm using a rooted phone with a custom ROM? If yes, then I can retry running the code after reverting the devices to the OEM builds (I'm not sure why this would cause an issue but custom ROMs can sometimes have missing implementations of features)

kirillzyusko, 2020-08-20:
> So you mean adding more log statements in react-native-wifi-p2p package?

Yes, it may be useful to see, where the errors occur. You can try to wrap the body of `getConnectionInfo` in `try/catch` statement and see, what error emerges there. I think it can be useful.

> And yes, I am never able to create group despite following the exact same order that was in your documentation. Can the issue here also be caused by the fact that I'm using a rooted phone with a [...]

It can be, however I'm not sure...

c-thun, 2020-08-28:
Tried both of your recommendations. Here is the attached log file. If you feel like, you may view my app code (I made the repo public) 

Here it is : https://github.com/c-thun/meridio

You will see two send cases in the adb logcat. I am trying to only send the message this time (regardless of the console output) and the first time, I send the name of one file as a message only while the other time I send the name of two files

[output_onceAgain.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5143513/output_onceAgain.txt)

And the logs attached below (output_twiceAgain.txt) is the logcat output when I just directly use ```sendMessage()``` as follows
```
 WiFiP2P.connect(macAddr)
        // Resolve connection details using getConnectionInfo
        .then(() => {
            console.log('Getting connection info');
            return WiFiP2P.getConnectionInfo().then(info => console.log('getConnectionInfo', info));
        })
        .then(() => {
            console.log('Sending Files');
            // return sendFiles(files);
            return WiFiP2P.sendMessage('Hey ya');
        })
        .catch(err => console.log('Error occurred', err));
```

And try to receive the message on the other device. I also tried this after reversing the roles to no avail

[output_twiceAgain.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5143533/output_twiceAgain.txt)

kirillzyusko, 2020-08-28:
I will try to run your repo on my phones.

c-thun, 2020-08-28:
By some random method : One device did manage to send a Hey message to the other (I saw it received on the second device) but that so far has just been an anomaly. Unable to reproduce it

kirillzyusko, 2020-09-01:
@c-thun Still haven't had a chance/free time to test it...
Too busy with main work, unfortunately :(

c-thun, 2020-09-01:
No worries. With Nearby Share being pushed out to all android users, my project no longer is time constrained and has become more of a enthusiast learning endeavour. Take your time. I am also between projects currently :)

kirillzyusko, 2020-09-01:
@c-thun By the way, did you consider to try to use [nearby](https://github.com/butchmarshall/react-native-google-nearby-connection) API?

c-thun, 2020-09-01:
I didn't know this API was there. This sounds perfect. I'll try this out asap. Thanks a  lot