kirillzyusko/react-native-wifi-p2p #43: react-native-wifi-p2p connect and send ZPL code to zebra printer via hf2211 wifi serial device server

Alnsentry, 2020-09-29:
Hi, my name is akbar, I have a project to create a mobile application for a zebra printer and I want the zebra mobile application and printer to connect with wifi and the mobile application can send zpl code to the printer via serial device server. can this library handle it?.

kirillzyusko, 2020-09-29:
Hey @Alnsentry
I didn't work with these devices so I can not assure anything. If you can connect to this printer over WiFi direct connection from Android settings and somehow control your device - then I would say yes, it's possible (but it may require some code modification for specific task).
But I can not say anything with 100% confidence in advance.

Alnsentry, 2020-09-29:
Thank you, for your response @kirillzyusko . I'll try first whether this works or not