kirillzyusko/react-native-wifi-p2p #44: {code:0, message: "Operation failed due to an internal error"}

prakash3720, 2020-12-24:
[image]
I am getting this error when I was tried the [react-native-wifi-p2p-example](https://github.com/kirillzyusko/react-native-wifi-p2p-example) without any code modification. I installed all the dependencies and executed 'npm run android' to run on my physical device.

kirillzyusko, 2020-12-25:
Hi @prakash3720 
Please, attach JS logs and adb logs as txt files
And what is the device and OS version do you use?

prakash3720, 2020-12-25:
> Hi @prakash3720
> Please, attach JS logs and adb logs as txt files
> And what is the device and OS version do you use?
> 
> P. S. Also it's worth to read #17

Hi @kirillzyusko 
Thank you so much for the quick response

This is the [log](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5741829/log.txt) file and I'm using OnePlus 5T which is running on android 10

I looked into #17 but couldn't come up with a solution.

kirillzyusko, 2020-12-25:
@prakash3720 yes, you are right - it's general error. You attached only JS logs. May I ask you to attach pure adb logs?

prakash3720, 2020-12-25:
> @prakash3720 yes, you are right - it's general error. You attached only JS logs. May I ask you to attach pure adb logs?

I'm getting the same log for adb log too. You mean the **adb logcat *:S ReactNative:V ReactNativeJS:V** command or is it anything else?

I executed the above command when the app was running in another terminal

kirillzyusko, 2020-12-25:
@prakash3720 only `adb logcat`. We may find something interesting if we grab all logs from device😉

prakash3720, 2020-12-26:
I'm getting [this](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5742750/adblog.txt) for adb logcat

prakash3720, 2020-12-26:
I'm getting [this](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5742781/logcat.txt) for adb logcat without any params and [this](https://github.com/kirillzyusko/react-native-wifi-p2p/files/5742782/adblog.txt) with params to filter out the logs of the app

kirillzyusko, 2020-12-26:
@prakash3720 I thinks it's the reason:
```
12-26 12:09:23.982  1563  1657 W BroadcastQueue: Permission Denial: receiving Intent { act=android.net.wifi.p2p.CONNECTION_STATE_CHANGE flg=0x24000010 (has extras) } to ProcessRecord{ed50ec8 11198:com.examples/u0a330} (pid=11198, uid=10330) requires android.permission.ACCESS_FINE_LOCATION due to sender android (uid 1000)
12-26 12:09:23.987  1563  1657 W BroadcastQueue: Permission Denial: receiving Intent { act=android.net.wifi.p2p.THIS_DEVICE_CHANGED flg=0x4000010 (has extras) } to ProcessRecord{ed50ec8 11198:com.examples/u0a330} (pid=11198, uid=10330) requires android.permission.ACCESS_FINE_LOCATION due to sender android (uid 1000)
12-26 12:09:24.000  1563  2556 E WifiPermissionsUtil: UID 10330 has no location permission
```

prakash3720, 2020-12-26:
@kirillzyusko That was my bad. I added `<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />` in AndroidManifest.xml but forgot to change `PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION` to `PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION` in App.js

This error is not coming up anymore. I'll close the issue now. Thank you so much!

closed 2020-12-26 by prakash3720