kirillzyusko/react-native-wifi-p2p #45: subscribeOnPeersUpdates doesn't fire at all it doesn't show me the connected devices Test on Android 10

Rananjaya, 2021-01-21:
subscribeOnPeersUpdates doesn't fire at all it doesn't show me the connected devices on Android 10

  async componentDidMount() {
      try {

          await initialize();
          // since it's required in Android >= 6.0
          const granted = await PermissionsAndroid.request(
            PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
              {
                  'title': 'Access to wi-fi P2P mode',
                  'message': 'ACCESS_COARSE_LOCATION'
              }
          )

          console.log(granted === PermissionsAndroid.RESULTS.GRANTED ? "You can use the p2p mode" : "Permission denied: p2p mode will not work");
           console.log("inside")
          subscribeOnPeersUpdates(this.handleNewPeers);
          subscribeOnConnectionInfoUpdates(this.handleNewInfo);
          subscribeOnThisDeviceChanged(this.handleThisDeviceChanged);

          const status = await startDiscoveringPeers();
          console.log('startDiscoveringPeers >>>>>>>>: ', status);
      } catch (e) {
          console.error("error >>>>",e);
      }
  }

kirillzyusko, 2021-01-21:
Hi @Rananjaya 
Did you follow [caveats)[https://github.com/kirillzyusko/react-native-wifi-p2p#caveats] instructions?

Rananjaya, 2021-01-25:
Hi @kirillzyusko Yes I followed  caveats instructions

kirillzyusko, 2021-01-25:
@Rananjaya could you attach logs from `adb logcat`?

But basically it's not a problem of this package. It's system code, so I think something is wrong with setup. So, please, provide logs :)

ShubhBhatt197, 2024-09-06:
Having same issue ,after    startDiscoveringPeers() function successfully started ,subscribeOnPeersUpdates does not return any thing

ShubhBhatt197, 2024-09-06:
@Rananjaya Have you solved this issue?