kirillzyusko/react-native-wifi-p2p #46: I need mesh topology

MuhammetAdamoglu, 2021-02-04:
I need a p2p connection between android devices, 3 different devices should be able to exchange data with each other, but without a server. Only clients should be able to communicate with each other. However, when I look at the document, we can only use the star topology. The mesh topology I need. How can I get this topology with this library?

kirillzyusko, 2021-02-04:
Hi @MuhammetAdamoglu 
Thanks for interesting question?
Is it possible to create mesh network using Android WiFi direct feature? Do you have small examples of such apps?

MuhammetAdamoglu, 2021-02-04:
Frankly, I cannot find many examples. I want to offer internet use with / without internet for my restaurant system. I need to create a network where cashiers, kitchens and waiters can communicate with each other. For example, when the waiter adds an order, it must be delivered to the kitchen and when the product is ready in the kitchen, he must inform the waiter that the order is ready. How can I create this network?

kirillzyusko, 2021-02-04:
@MuhammetAdamoglu yes, you described mesh network. Have you considered the usage of [nearby](https://github.com/butchmarshall/react-native-google-nearby-connection) API? Can it be done through it?

MuhammetAdamoglu, 2021-02-05:
I reviewed the Nearby api, the library was updated 3 years ago and I think it doesn't work on current android devices. I will try to establish a socket connection in the local network with [this library](https://www.npmjs.com/package/react-native-static-server). I hope it works.

As far as I know, the p2p connection must be based on the mesh topology. I know wrong?

kirillzyusko, 2021-02-05:
@MuhammetAdamoglu not sure I get your question. Could you elaborate a little bit more?🙂

Regarding first part, do you think react-native-static-server will allow p2p communication? As I could understand from the description it may serve only static files (not dynamic content, like, for example, text). Or I'm missing something?

kirillzyusko, 2023-03-15:
Closed since https://github.com/kirillzyusko/react-native-wifi-p2p/pull/61 was merged

closed 2023-03-15 by kirillzyusko