kirillzyusko/react-native-wifi-p2p #47: Sharing App using same network

muzammil413tariq, 2021-04-29:
Hey i want to create an app using react native whixh can share data from mobile to desktop

kirillzyusko, 2021-04-29:
@muzammil413tariq Does your desktop support android p2p-protocol?

muzammil413tariq, 2021-04-29:
@kirillzyusko  let me know how to check it ...and if its support then is this is possible to send data from mobile to desktop

kirillzyusko, 2021-04-29:
@muzammil413tariq how do you develop desktop application? Which framework are you using?

muzammil413tariq, 2021-04-29:
@kirillzyusko  Electron js

kirillzyusko, 2021-04-29:
@muzammil413tariq I don't think you can use this library for these purposes. I quickly ran through several links in Google and couldn't find any info on how to connect mobile device with PC.

muzammil413tariq, 2021-04-29:
@kirillzyusko so how can i send data from my mobile react native app  to desktop..Like shairit.

muzammil413tariq, 2021-04-29:
Many of modern device has in built support for this... I creating an react native app in which user can click images and send it to desktop..please suggest m how can i do it

kirillzyusko, 2021-04-29:
@muzammil413tariq if you can give me an example (source code) of app, that can share files between desktop/mobile, then I can help.

I don't know almost anything about development desktop applications.

muzammil413tariq, 2021-04-30:
@kirillzyusko  here is what I found on the internet is #react-native-tcp I think this can help me but i don't know  how to use it

kirillzyusko, 2021-04-30:
@muzammil413tariq you can try. But I don't know how you can establish connection between desktop and mobile. Sorry, I can't really help you...