kirillzyusko/react-native-wifi-p2p #54: unsubscribeFromPeersUpdates : EventEmitter.removeListener() : Method has been deprecated

seba9999, 2022-10-25:
Hi, I'm using this lib (well at least trying to ^^) in a Expo project.

I was wondering if you're planning to change the `removeListener()` Method as adviced : 

```
 WARN  EventEmitter.removeListener('WIFI_P2P:PEERS_UPDATED', ...): Method has been deprecated. Please instead use `remove()` on the subscription returned by `EventEmitter.addListener`.
```

It happends only when I try to use `unsubscribeFromPeersUpdates()` on unmount.

kirillzyusko, 2022-10-25:
Hi @seba9999 
Yes, I will try to fix it on this week. Most likely it will require to change API of library a little bit, but I will try to fix it and make a new release this week.

seba9999, 2022-10-25:
Thanks a lot ! 

Could you elaborate on how can I create a broadcast system with your lib please ? :) 

If I have 3 devices (A, B and C) and they need to have some up to date datas so as to let's say a group chat.

A send "Hello"
B send "World"
A, B and C should be able to write :
- A : Hello
- B : World

And here is my main question : (the sendMessage Doc)[https://github.com/kirillzyusko/react-native-wifi-p2p#sendmessagemessage] isn't clear with multiple devices.

If A creates a group and B and C join it ... 
Will a message sent by A be received from B and C ?
Will a message sent by B or C be received from A ? (sur... ?) But what about C ?

kirillzyusko, 2022-10-26:
@seba9999 I think I can't help here a lot, because I've never had 3 working Android devices at the same time (so I haven't tested this case). The only thing I can do is to recommend to read official documentation about wifi-p2p connection on Android website :(

Theoretically you need to organize star-topology network, where in the center device will act as a group owner and will broadcast events which arrived from a single device to all others.

But I may assume you'll need to change java code of this library to achieve this. In any cases - PRs are welcome 😊

closed 2022-10-26 by kirillzyusko

seba9999, 2022-10-26:
I've never tried such a task ... Helping in java to develop an RN lib ... 
But why not give it a try ... 

Could you address me a starting point ? ... So as how to even build your code ? And how to test it ... Is it even possible using expo ? Or I'll need android studio and some sandbox android app in java ?

I don't even know where to start ^^'

kirillzyusko, 2022-10-26:
@seba9999 yes, I'm using Android Studio for package development. I tested implementation using this test app: https://github.com/kirillzyusko/react-native-wifi-p2p-example (but I used symlinking)

So first of all you'll need to clone test app, then clone this repo and then using `symlink` you'll need to substitute a package from `npm` (in `node_modules`) to a locally cloned package. Then open example project in Android studio:

[image]

And that's all 🎉 All java code modification will be reflected in original library repo (thanks to symlinking).

If something is not clear about the workflow - try to ask and I will answer, but it would be better to open a new issue in order to avoid information chaos 😅