kirillzyusko/react-native-wifi-p2p #56: Adding something like sendMultipleMessages(message, addresses, promise)

seba9999, 2022-10-27:
Hi it's me again ! 
 
 I've tried to read your code and I think I understood like 40% ... But I've read carefully the sendMessage workflow and I was wondering if creating another method sendMessage with another parameter : A map of address where the message should be sent.
 
 
 I would imagine something like this : 
 
 1 - Create another Method : `sendMessageTo(message, address)` which is almost the same as the current one except that you'll change [this line](https://github.com/kirillzyusko/react-native-wifi-p2p/blob/master/android/src/main/java/io/wifi/p2p/WiFiP2PManagerModule.java#:~:text=serviceIntent.putExtra(MessageTransferService.EXTRAS_GROUP_OWNER_ADDRESS%2C%20wifiP2pInfo.groupOwnerAddress.getHostAddress())%3B) : 

    serviceIntent.putExtra(MessageTransferService.EXTRAS_GROUP_OWNER_ADDRESS, wifiP2pInfo.groupOwnerAddress.getHostAddress());

to

    serviceIntent.putExtra(MessageTransferService.EXTRAS_GROUP_OWNER_ADDRESS, address);
 
 
And modify the ResultReceiver with something like that : 
 
    @Override
    protected void onReceiveResult(int resultCode, Bundle resultData) {
        if (resultCode == 0) { // successful transfer
            return true;
        } else { // error
           return false;
        }
    }
 
 2 - Create the `sendMultipleMessages(message, addresses, promise) ` and loop over addresses calling the previously created `sendMessageTo`
 
     public void sendMultipleMessages(String message, String[] addresses, final Promise promise) {
            for (String address : addresses){
                sendMessageTo(message, address);
            }
     }
     
Then get if there's a falsy return to reject the promise or if all are true and the message actually was send to all the addresses ...

What do you think about that ? Sorry my java is a bit rusty ^^'

kirillzyusko, 2022-10-27:
Hi @seba9999 
I think it's working method 🙂 One thing that I would do potentially differently is an iteration - can we iterate in JS code?

Just theoretically we can slightly change internals of the library and change the signature of `sendMessage`:
```diff
-public void sendMessage(String message, final Promise promise) {
+public void sendMessageTo(String message, String address, final Promise promise) {
+public void sendMessage(String message, final Promise promise) { // just to keep backward compatibility
+  this.sendMessageTo(message, wifiP2pInfo.groupOwnerAddress.getHostAddress(), promise)
+}
```

Then I think `broadcastTo` method can be fully implemented on JS side (through `Promise.all` construction). What do you think?

I would love to help with the implementation, but I don't have even 2 physical devices.

seba9999, 2022-10-28:
What do you mean by iterate ? Looping through a list of adresses ? Of course we could do it in JS ! The real question is in which side we want to handle the potentials errors  / non-receive...  And I'm afraid I don't know where's the best place place to do so ( JS / java )

Btw both should work ! 

I would be very happy to help to test it out ! I've got 3 devices ( thanks family :D ) to test it out. If you make a branch with such modifications I would import the new branch and test it on my app for you no problem 👍

kirillzyusko, 2022-10-28:
> What do you mean by iterate ? Looping through a list of adresses ?

Yes, I meant this.

> I'm also wondering (that's a part I didn't fully understood) If the "port" setting should be changed or not ?

Good question :) I think the answer could be discovered only by trial and error 😄 

> I would be very happy to help to test it out ! I've got 3 devices ( thanks family :D ) to test it out. If you make a branch with such modifications I would import the new branch and test it on my app for you no problem 👍

I will try to find a time on this weekend and modify code. But can not assure the code will work 🙈

seba9999, 2022-11-01:
Let me know when there's something to try :)

seba9999, 2022-11-03:
I've found a fork from your project that provide 2 other functions linked to services ... 

https://github.com/m3tasploit/react-native-wifi-p2p-reborn

Would you mind having a look at [the changes](https://github.com/kirillzyusko/react-native-wifi-p2p/compare/master...m3tasploit:react-native-wifi-p2p-reborn:master) ? I'm not 100% sur it create some broadcast feature ... the [Android wifip2pManager Doc ](https://developer.android.com/reference/android/net/wifi/p2p/WifiP2pManager) isn't clear enough IMO.

kirillzyusko, 2022-11-04:
Hi @seba9999 

I think `reborn` version adds TS support and a wrapper around `WifiP2pDnsSdServiceInfo` [class](https://developer.android.com/reference/android/net/wifi/p2p/nsd/WifiP2pDnsSdServiceInfo).

I don't have experience with this stuff, to be honest. After a quick search in a google for me it seems like it's needed for device management (like printers) that are connected to the same network (though I may be mistaken and I may assume that it's more powerful stuff than I suppose 🤷‍♂️ ).

As you pointed out Android documentation is not clear especially for p2p stuff. For me it seems like `DnsSd` is kind of a protocol that is needed for specific use cases. But once again, I'm not an expert in this :(

mateomarconi, 2023-01-21:
Hi, @seba9999 @kirillzyusko @

I am looking for some kind of broadcast too. I just need that A, B and C shares the same information the app is handling at the same time, as A and C are not connected (because wifi restrictions on distance), C would't receive the data that A has modified. Another use case it would be that B takes some space from the other users (physically speaking) and get disconnected from A or C (BUT never between both), also A and C get connection stablished (or maybe not).

Could these examples help to achieve broadcast between all connected devices in a group?

Sends data to all connected devices in the group. This function should iterate through the list of connected devices and send the data to each one individually.

```
function broadcastData(data) {
  wifi.getConnectedPeers((peers) => {
    peers.forEach((peer) => {
      wifi.sendData(peer.address, data);
    });
  });
}
```

Listen for incoming data, and check if the data is intended for the broadcast.

```
wifi.onDataReceived((data) => {
  if (data.isBroadcast) {
    // Handle broadcast data
  } else {
    // Handle regular data
  }
});
```

In order to know that the data is intended for the broadcast, add a flag or a specific key-value pair in the data. For example, add the key isBroadcast with the value true in the data before sending it.

```
const data = {
  message: 'Hello World',
  isBroadcast: true
};
broadcastData(data);
```

seba9999, 2023-03-13:
Hi @kirillzyusko,

It's been a while and I wasn't able to work on this since November but I would like to work on this again. Are you planning to try to implement the broadcast thing we were talking about ? :) 

@mateomarconi 

Did you achieve somehting like you described ?

viniciuscb, 2023-03-14:
Hi. I am working in a change that solves this issue and also #57 . 

I had trouble compiling the code of react-native-wifi-p2p (maybe some issue with gradle) and tried to implement it using react-native-wifi-p2p-reborn.

Implemented here sendFileTo (untested) and sendMessageTo (testing in my application).

There is a **breaking change** in receiveMessage(), that now instead of receiving just the message (string), it receives an object with the following structure: 

```
{
  fromAddress: "192.168.43.20",
  message: "message received"
}
```

I had to do that to know the IP of the nodes in the network, besides the group owner. So, in my application logic, after the connection is estabilished, the group owner starts to receive messages (using receiveMessage() ) and the nodes send a first message.

I think it is possible now to cherry-pick the changes in this commit to this project. If approved, the README also will have to be changed.

kirillzyusko, 2023-03-14:
Hi @viniciuscb 

Would you mind to submit a PR? Your changes looks good for me 👍

I'm compiling the project using test example project: https://github.com/kirillzyusko/react-native-wifi-p2p-example

There you can specify a package from github:

```json
"react-native-wifi-p2p": "https://github.com/seba9999/react-native-wifi-p2p#feature-broadcast"
```

Or you can use [symlink](https://apple.stackexchange.com/a/115647) to link a local package (I prefer this option, since I can easily modify file locally and I don' need to reinstall deps every time).

@viniciuscb just want to ask - did you test your changes? Do they actually work? If yes, then I'd like to ask you to add info about new methods in README and submit a PR 😊 I'd love to merge them and release a new version of the library!

kirillzyusko, 2023-03-15:
Closed since https://github.com/kirillzyusko/react-native-wifi-p2p/pull/61 was merged

Thanks @viniciuscb and @seba9999 for contributing to the project! I really appreaciate your help! 😊

closed 2023-03-15 by kirillzyusko

viniciuscb, 2023-03-15:
Hi. @kirillzyusko . Nice to contribute.

mateomarconi, 2023-08-02:
Thank you @viniciuscb! I will try it out soon!

@seba9999