kirillzyusko/react-native-wifi-p2p #57: Urgent: Sending File from Server to Client

Rishy20, 2022-11-12:
Hi, I have created a react native app where initially the server creates a group, and then the client connects to the group. Then the server initiates receiveMessage() and the client sends the name of the files required through sendMessage(). The server should send those files to the client. I tried sending files by calling receiveFiles() in the client and then sendFiles() in the server. In the server it displays as "Files sent successfully" but the client doesn't receive the files. How can I send files from the server to the client to cater to the above requirement?

kirillzyusko, 2022-11-12:
Hi @Rishy20 

If you send a single file from server to client - does it work or not? The reason why I'm asking is to understand whether a problem with sending multiple files or with sending files from server to client.

Rishy20, 2022-11-12:
Hi @kirillzyusko 
I tried sending a single file. It didn't work. So I hope the problem is sending from server to client. Is it because when sending from server it doesn't point to client's address?

kirillzyusko, 2022-11-13:
@Rishy20 well, honestly I haven't tried to send files from server to client. So I assume it may not work.
Unfortunately I don't have two physical devices right now, so I can not verify why it doesn't work :(

kirillzyusko, 2023-03-15:
Closed since https://github.com/kirillzyusko/react-native-wifi-p2p/pull/61 was merged

closed 2023-03-15 by kirillzyusko