kirillzyusko/react-native-wifi-p2p #58: Could not resolve all files for configuration ':react-native-wifi-p2p:classpath'

aminisaeed, 2022-11-12:
hi
it install without problem 
but when i want to run on simulator this error apear:
Could not resolve all files for configuration ':react-native-wifi-p2p:classpath'. > Could not find builder-3.3.0.jar (com.android.tools.build:builder:3.3.0).
Searched in the following locations:
         https://dl.google.com/dl/android/maven2/com/android/tools/build/builder/3.3.0/builder-3.3.0.jar

when i check the "Could not find builder-3.3.0.jar" address it was 404 not found
how should i solve this problem?
I need this library

kirillzyusko, 2022-11-12:
Hi @aminisaeed 
I clicked a little nk and download has begun. Maybe something is wrong with your network configuration and it can not download because the resource is blocked?
Or maybe the page was under temporary maintenance?
What happens if you follow the link?

aminisaeed, 2022-11-12:
[image]

 result for me

kirillzyusko, 2022-11-13:
@aminisaeed weird, but when I click on it - download begins 🤷‍♂️ 

[image]

aminisaeed, 2022-11-14:
The problem was from my VPN
I can download it with chrome but the speed was very slow

Now I have that file (30Mb) and again in the react native run android that error appears, and can't download this file

where should I put that file(builder.3.3.0.jar    30Mb) ? That in the installation react detect it and use it instead of download it?
In which folder???

kirillzyusko, 2022-11-14:
@aminisaeed could you try this? https://stackoverflow.com/a/10834567/9272042

aminisaeed, 2022-11-15:
The problem solved
The error was from my VPN and connection

kirillzyusko, 2022-11-15:
Awesome! I'm closing the issue.

closed 2022-11-15 by kirillzyusko