kirillzyusko/react-native-wifi-p2p #59: Android 13  rejects startDiscoveringPeers() with code 0

verybluebot, 2023-02-06:
Requested `ACCESS_COARSE_LOCATION` and `ACCESS_FINE_LOCATION` 
in AndroidManifest.xml 
and I also ask for user permission dynamically at run time.

Went over issues [#13](https://github.com/kirillzyusko/react-native-wifi-p2p/issues/13)
and [#17](https://github.com/kirillzyusko/react-native-wifi-p2p/issues/17)

Seems not my case, as I state and request the needed permissions and 
Always get `0` error code on `startDiscoveringPeers()`.

Also the same code is working perfectly on older android OS (< 13).

Stumbled on this, but not sure I fully understand the hardware requirements/ compatibility issue this mite rise:
[from Android docs](https://source.android.com/docs/core/connect/wifi-direct#examples_and_source) :

"
To use this feature, device manufacturers must implement the Wi-Fi Vendor and Supplicant HAL interfaces. Starting with Android 13, the Supplicant interface uses AIDL for the HAL definition. For releases before Android 13, interfaces and vendor partitions use HIDL.
"

Stack: 
RN wifi p2p: 3.3.0
RN: 0.62.2
React: 16.11.0

Any ideas?

verybluebot, 2023-02-06:
Ok I missed `android.permission.NEARBY_WIFI_DEVICES` permission that needs to be added to a AndroidManifest.xml 
`<uses-permission android:name="android.permission.NEARBY_WIFI_DEVICES" />`

Not sure it was mentioned in this repo. 

And then request it dynamically in runtime.

Im using `react-native-permissions` and had an older version which does not supported `NEARBY_WIFI_DEVICES`
So if you are in the same use case upgrade to latest version and where support for that permission was added.

Leaving this open in case it will help some one else

kirillzyusko, 2023-02-06:
Hi @verybluebot 
Thanks for opening the issue.

Would you mind to submit a PR with updated steps (i.e. new permission and dynamic permission request via `react-native-permissions`.

I think it will be helpful for other people to get this information straight from README 🙂

kirillzyusko, 2023-03-16:
Closed because https://github.com/kirillzyusko/react-native-wifi-p2p/pull/70 and https://github.com/kirillzyusko/react-native-wifi-p2p/pull/69 were merged

closed 2023-03-16 by kirillzyusko

Luo-XD, 2023-04-24:
I also had this problem with **Android 13**.
Refer to the treatment of problem #44 and #59 .
`NEARBY_WIFI_DEVICES`, `ACCESS_FINE_LOCATION`, and `ACCESS_COARSE_LOCATION` must be met simultaneously：

- Add **dependency**
`yarn add react-native-permissions`
-  **AndroidManifest.xml** permission
`+ android.permission.NEARBY_WIFI_DEVICES`
`+ android.permission.ACCESS_FINE_LOCATION`

- **app.js**  permission code
`initialize()
      .then(() => PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION))
      .then(() => PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION))
      .then(() => PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.NEARBY_WIFI_DEVICES))
      .catch((err: any) => console.log('initialization was failed. Err: ', err));`

I spent a lot of time, and I hope it helped！

kirillzyusko, 2023-04-24:
@Luo-XD did you add `ACCESS_COARSE_LOCATION` in AndroidManifest?

Also it would be very helpful if you could attach `adb logs` 🙂

steve-lgtm, 2024-03-01:
> I also had this problem with **Android 13**. Refer to the treatment of problem #44 and #59 . `NEARBY_WIFI_DEVICES`, `ACCESS_FINE_LOCATION`, and `ACCESS_COARSE_LOCATION` must be met simultaneously： [...]

this helps me a lot. I'm on Android14

thanks

kirillzyusko, 2024-03-06:
@steve-lgtm would you mind to submit a PR with adding this information to README? 👀