kirillzyusko/react-native-wifi-p2p #66: sendMessage and receiveMessage refactoring?

viniciuscb, 2023-03-15:
I am having another problem now, regarding the sendMessage method. Sometimes the connection is closed and I could not have this information before sending another message. This kills the app. [Here](https://github.com/kirillzyusko/react-native-wifi-p2p/blob/562c5496e160a96a866e30eb9fc343e7732f449e/android/src/main/java/io/wifi/p2p/WiFiP2PManagerModule.java#L334) is the problem, because groupOwnerAddress is null and I cannot call getHostAddress() on null.

Besides that, I think there should be some way to stop receiving messages. If I call receiveMessage() and (unexpectedly) don't receive any message, a port will stay open waiting for messages. This is a real issue when I call it in a screen, go out and enter this screen again.

Maybe adding a stopReceivingMessage() method or replacing for two methods, startReceivingMessages() and stopReceivingMessages()

kirillzyusko, 2023-03-16:
Hi @viniciuscb 

I think it make sense to add `stopReceivingMessage` and `stopReceivingFile` 🙂 Would you mind to contribute and add these methods? 👀 

> Sometimes the connection is closed 

Would be good to understand the reason behind it, i. e. why the connection was closed. Is it a bug in a library or it depends on external factors, such "user closed app"/"distance to receiver side became too big and because of that the signal was lost"?
Also maybe would be good to send an event "CONNECTION_CLOSED" or something like that, if we actuall can derive this event.

viniciuscb, 2023-03-16:
Hi Kirill. The problem with sendMessage is happening due to external factors. For instance,  when the app crashes/is closed in one device, and I didn't receive yet the information that the connection changed in react native level (through listeners/subscribe methods).

kirillzyusko, 2023-03-16:
Got it @viniciuscb 👍 
Then I think we'll need to have `stopReceivingMessage`/`stopReceivingFile` methods and it would be very helpful to know, when "CONNECTION_CLOSED" event occured (to actually call `stopReceivingMessage`/`stopReceivingFile` methods). Agree?

viniciuscb, 2023-03-16:
I see now that the AsyncTask class became deprecated in API 30. So the receiveMessage will need to have a refactoring anyway.

https://developer.android.com/reference/android/os/AsyncTask

Besides that, @kirillzyusko and @seba9999 , maybe we can pass an optional parameter to receiveMessage to receive also the metadata (currently only the origin IP address). This way we can maintain backward compatibility.

kirillzyusko, 2023-03-17:
@viniciuscb I think if you rewrite and migrate away from AsynTask, then you may not worry about backward compatibility :)

In my opinion it's useful to know sender address - of course it will be breaking change, but it will be easy to migrate. Moreover I already have some deprecated methods in API and I would like to remove them and publish new 4.0.0 version :)

viniciuscb, 2023-03-20:
I've proposed a PR ( #80 ) that does not break backward compatibility.

PR #80 (2023-03-24): Refactor in receiveMessage
* receiveMessage now uses Executor, instead of the deprecated AsyncTask;
* stopReceivingMessage() added: closes the listening port if no message has been received
* a new optional parameter has been added to the receiveMessage method. If receiveMessage receives an object {meta: true}, then the promise result is not a string, but an object with the following format:

  { fromAddress: "<ip address of the origin in the p2p wifi network>", message: "original message (string)" }
  
  
This fixes #66 .

@seba9999 can you please help me test this modification? I've tried he [... cut]
closed 2023-03-24 by kirillzyusko