kirillzyusko/react-native-wifi-p2p #77: SendMessageTo - can't use MAC address and we don't have ip addresses

seba9999, 2023-03-18:
The list of peer only provide peers info with some MAC addresses ... But for now the sendMessage uses `InetSocketAddress` which can't use MAC addresses.

I'll maybe try to pass the ip from the client to the server at first message ( in order to be able to transfert message from group owner to clients )

Or we need some refactoring in the sendMessage function to add extra fields to store ip list of devices / peer in the group ...

kirillzyusko, 2023-03-20:
> I'll maybe try to pass the ip from the client to the server at first message ( in order to be able to transfert message from group owner to clients )

Hi @seba9999 
Let me know, whether it works for you or not 👀

viniciuscb, 2023-03-20:
Hi @seba9999 . I tried to find how can one know the ip addresses of other peers. What I've found is that people is sending a first message from clients (peers) to group owner to know the IP. After that, the communication continues normally.

- https://stackoverflow.com/a/33405178/1216027
- https://stackoverflow.com/a/14525688/1216027
- https://stackoverflow.com/a/10126598/1216027

I think that the README should be updated explaining this usage.

viniciuscb, 2023-03-20:
Hi @seba9999 . The problem is that the information that we receive in react native is exactly the information that android gives us. 

I've sent a PR now ( #80 )  that changes the receiveMessage method, and we also receive in react native the ip of the sender. In my application I am using like this:

```javascript 

  if (connectionInfo.groupFormed) {
    if (!connectionInfo.isGroupOwner) {
      sendMessage('it can be anything here');
    } else {
      receiveMessage({meta: true}).then(({fromAddress, message}) => {
        // now you have the address of the peer in fromAddress. You can store it
        // somewhere to use in the future: sendMessageTo(message, address)
      });
    }
  }

```

viniciuscb, 2023-03-20:
I've not tried to send a message to the broadcast address, though.

viniciuscb, 2023-03-20:
as receiveMessage receives just one message, I'm implementing it the following way:

```javascript

  const onReceiveMessage = () => {
    receiveMessage({meta: true})
      .then({ fromAddress, message } => {
        let receiveNextMessage = true;

        // Conditions: depending on the message received or other logic, will 
        // close the connection and set receiveNextMessage to false;

        if (receiveNextMessage) {
          onReceiveMessage();
        }
      })
      .catch(err => {
        console.log(user.name + ':   Error while message receiving', err);
        onReceiveMessage();
      });
  };

```

Suyashbajpai27, 2024-03-15:
I have same issue that i am not able to send message because it is capturing the MAC address not IP address.