kirillzyusko/react-native-wifi-p2p #81: sendFile and receiveFile refactoring

viniciuscb, 2023-03-24:
Do the same refactoring as it was done with sendMessage and receiveMessage.

viniciuscb, 2023-03-24:
@kirillzyusko what do you think about changing the socket listening to files to port 8989, instead of the port 8988? We are using 8988 for message already...

(of course this would break backward compatibility, as peers using old library versions could not send files to peers with new library versions....)

viniciuscb, 2023-03-24:
I think we should update the documentation. For instance, the sendMessage() instructions in readme worked for me a bit differently. 

It is not necessary anymore to create a group / be the owner of the group to send and receive messages. More than that, here in my tests (and the tests of my team) it was not possible to create a group (the android system gives me a "resource busy" message) and not even start as group owner with connectWithConfig . 

It seems that a groupOwner is chosen randomly by android, but it haves some criteria: here everytime my newest mobile device was elected to be the groupOwner, the same for tests with my teammates.

seba9999, 2023-05-29:
Hey folks ! 
I've been busy the last few monthes but I'm planning on trying again broadcast with RN !

@viniciuscb You're saying that it's not necessary to create a group anymore ... But how do you manage to have the "first" adress to send message then ? Which actions are mandatory to connect 3 (or more) devices and then each user could brodcast to everybody ? 

Would you be kind enough to share some bootstrap code ? 😇

kirillzyusko, 2023-05-30:
@seba9999 I can create v4 if needed - don't be afraid to make breaking changes, but also be sure that you know steps for migration from v3 to v4 🙂