kirillzyusko/react-native-wifi-p2p #83: I want to make a real-time-streaming (Two Android devices connected 1:1)

munsangu, 2023-03-29:
Hello, I am using a library for connecting between Android devices with React native. (Android 13 version) 
I succeeded in connecting the two devices, but I want to know if real-time-streaming is also possible with this library.
(This means that B Android simultaneously watches the video that A Android is shooting)

Thank you, very much

kirillzyusko, 2023-03-29:
@munsangu I highly doubt it's possible. Just theoretically you can send files picture by picture, but I don't think it will give a stable framerate. This library doesn't support streaming (yet) and in order to add it you'll need to modify source code.

Is there any apps which are using wi-fi direct for streaming?

munsangu, 2023-03-29:
@kirillzyusko Thank you for answer. I'm making an android app with react-native, but after seeing that react-native-multipeer doesn't support streaming, I was looking for various libraries and trying to use what you made. There are two functions I'm trying to create. 
1. WiFi or Bluetooth connection between two Android devices
2. Watching the video being shot by one Android device on the other Android device in real time
3. The observing Android controls the connected Android device.
e.g) Android Phone + Robot vacuum cleaner with camera 
ㄴ Android Phone can see the video being filmed by the robot vacuum cleaner and can control moving direction of the robot vacuum cleaner

I'm not an android developer, so it's more difficult. 😭

munsangu, 2023-03-30:
@kirillzyusko 
Thank you for caring me. The project manager said he would do it himself.😅