kirillzyusko/react-native-wifi-p2p #85: Get Available devices not working

mariahourani, 2023-07-03:
import React, { useEffect, useState } from 'react';
import {
  StyleSheet,
  View,
  Button,
  Alert,
} from 'react-native';
import { WiFiP2PManager } from 'react-native-wifi-p2p';

import {
  initialize,
  startDiscoveringPeers,
  stopDiscoveringPeers,
  subscribeOnConnectionInfoUpdates,
  subscribeOnThisDeviceChanged,
  subscribeOnPeersUpdates,
  connect,
  cancelConnect,
  createGroup,
  removeGroup,
  getAvailablePeers,
  sendFile,
  receiveFile,
  getConnectionInfo,
  getGroupInfo,
  receiveMessage,
  sendMessage,
} from 'react-native-wifi-p2p';
import { PermissionsAndroid } from 'react-native';

const App = () => {
  const [devices, setDevices] = useState([]);

  useEffect(() => {
    
    const initializeP2P = async () => {
      try {
        await initialize();
        const granted = await PermissionsAndroid.request(
          PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
          {
            'title': 'Access to Wi-Fi P2P mode',
            'message': 'ACCESS_COARSE_LOCATION'
          }
        );

        console.log(granted === PermissionsAndroid.RESULTS.GRANTED ? "You can use the P2P mode" : "Permission denied: P2P mode will not work");

        console.log("DISCOVERY BEGUN");

        const peersUpdatesSubscription = subscribeOnPeersUpdates(handleNewPeers);
        const connectionInfoUpdatesSubscription = subscribeOnConnectionInfoUpdates(handleNewInfo);
        const thisDeviceChangedSubscription = subscribeOnThisDeviceChanged(handleThisDeviceChanged);

        const status = await startDiscoveringPeers();
        console.log('startDiscoveringPeers status: ', status);
        onGetAvailableDevices();

        return () => {
          peersUpdatesSubscription.remove();
          connectionInfoUpdatesSubscription.remove();
          thisDeviceChangedSubscription.remove();
        };
      } catch (e) {
        console.error(e);
      }
    };

    initializeP2P();
    
    const handleNewPeers = ({ devices }) => {
      console.log('OnPeersUpdated', devices);
      const updatedDevices = devices.map(device => ({
        ...device,
        name: device.deviceName // Add the device's name to the device object
      }));
      setDevices(updatedDevices);
  
      // Call getAvailablePeers() after updating the devices list
      getAvailablePeers()
        .then(peers => {
          console.log(peers);
          setDevices(peers);
        })
        .catch(err => console.error('Error while getting available devices', err));
    };

    
    const handleNewInfo = (info) => {
      console.log('OnConnectionInfoUpdated', info);
    };
  
    
    const handleThisDeviceChanged = (groupInfo) => {
      console.log('THIS_DEVICE_CHANGED_ACTION', groupInfo);
    };
  
    
  }, []);

  const connectToFirstDevice = () => {
      console.log('Connect to: ', devices[0]);
      connect(devices[0].deviceAddress)
        .then(() => console.log('Successfully connected'))
        .catch(err => console.error('Something gone wrong. Details: ', err));
  };

  const onCancelConnect = () => {
    cancelConnect()
      .then(() => console.log('cancelConnect', 'Connection successfully canceled'))
      .catch(err => console.error('cancelConnect', 'Something gone wrong. Details: ', err));
  };

  const onCreateGroup = () => {
    createGroup()
      .then(() => console.log('Group created successfully!'))
      .catch(err => console.error('Something gone wrong. Details: ', err));
  };

  const onRemoveGroup = () => {
    removeGroup()
      .then(() => console.log('Currently you don\'t belong to a group!'))
      .catch(err => console.error('Something gone wrong. Details: ', err));
  };

  const onStopInvestigation = () => {
    stopDiscoveringPeers()
      .then(() => console.log('Stopping discovery was successful'))
      .catch(err => console.error(`Something went wrong. Maybe your Wi-Fi is disabled? Error details`, err));
  };

  const onStartInvestigate = () => {
    startDiscoveringPeers()
      .then(status => console.log(`Status of discovering peers: ${status}`))
      .catch(err => console.error(`Something went wrong. Maybe your Wi-Fi is disabled? Error details: ${err}`));
  };

mariahourani, 2023-07-03:
I've been trying to fix this for a week, I have 3 phones connected thru wifi direct but when I try get available devices, it returns an empty array of devices

kirillzyusko, 2023-07-03:
@mariahourani could you attach `adb logcat` output as well? Just for period, when you are trying to request all available peers?

vipulav-j, 2023-11-22:
I am also getting same empty list.

kirillzyusko, 2023-11-22:
@vipulav-j please, attach you `adb logcat` logs. Without them I'll not be able to help 😔