kirillzyusko/react-native-wifi-p2p #86: getAvailablePeers() always return empty array

ForamPattha, 2023-07-12:
const initialiseP2P = async () => {
    const init = await initialize()
    console.log('In initialiseP2P', init);
    // since it's required in Android >= 6.0
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
      {
        title: 'Access to wi-fi P2P mode',
        message: 'ACCESS_FINE_LOCATION',
      },
    );
    if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      onStartDiscoveringPeers()
    } else {
      Alert.alert('Warning', 'Permission denied: p2p mode will not work')
    }
  };

const onStartDiscoveringPeers = async () => {
    await startDiscoveringPeers()
      .then(() => getAvailablePeers())
      .then(({devices}) =>
        console.log('Available Devices:', devices),
     
);
  };

Use above code but not getting near by device list ..I have tested this with 2 to 3 android device which Android version is 10 or up. Any help would be appreciated...thanks in advance.

kirillzyusko, 2023-07-12:
@ForamPattha please, attach `logcat` output just for period of time when you've tried to perform your actions. I can't help just based on the code you've attached 😔

ForamPattha, 2023-07-13:
@kirillzyusko can you  please check this attach ss . 
[image]

Available Devices: []

kirillzyusko, 2023-07-13:
@ForamPattha You've attached logs only for 100ms... Can you attach from time when you initialize module till the time when it returns an empty array?

ForamPattha, 2023-07-13:
@kirillzyusko thanks now it's working but I got following when use sendMessage().I want to some confirmation from receiver side and on behalf of confirmation I need to visible 1 button for that I use sendMessage() method but it generated following error  **Attempt to read from field 'java.net.InetAddress android.net.wifi.p2p.WifiP2pInfo.groupOwnerAddress' on a null object reference at io.wifi.p2p.WiFiP2PManagerModule.sendMessage(WiFiP2PManagerModule.java:339)**

Use of sendMessage()
---------
sendMessage("Ready To Receive")
        .then((metaInfo) => console.log('Message sent successfully', metaInfo))
        .catch(err => console.log('Error while message sending', err));