kirillzyusko/react-native-wifi-p2p #95: startDiscoveringPeers function is working in debug mode but not working in release apk anyone has any solution about that?

Suyashbajpai27, 2024-03-15:
I am trying to connect two android devices for sending data from one device to other device using this library but in debug mode it is working to search Nearby Devices but when i create a release build or apk ,then it is not able to search any nearby devices.

kirillzyusko, 2024-03-15:
@Suyashbajpai27 do you see any error in `logcat` output? Maybe there will be a useful hints what exactly blocks peers from being discovered?

Suyashbajpai27, 2024-03-15:
No ,because it is working in debug mode , i have no clue about that in
release mode.

kirillzyusko, 2024-03-15:
@Suyashbajpai27 you can spin up a logcat instance by `adb logcat` command or simply open it in Android Studio 🙂  And check your logs (it should contain all logs).

I'm more than sure that Android will log a root cause of why discovery fails. Feel free to try 🙌

Suyashbajpai27, 2024-03-15:
No , I didn't get any unusual logs in adb logcat in android studio.

kirillzyusko, 2024-03-15:
@Suyashbajpai27 can you share them here? (Just for 5-10 seconds, when you started to discover peers and got a particular status)

Suyashbajpai27, 2024-03-15:
[AdbLogcat.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/14620578/AdbLogcat.txt)
[AdbLogcat.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/14620592/AdbLogcat.txt)

Suyashbajpai27, 2024-03-15:
Hey! this library is working in Android version 11 or 12 or older version of android but not in 13.check the issue please.

kirillzyusko, 2024-03-16:
@Suyashbajpai27 according to yours logs I see that device actually can be found:

```
03-16 01:51:59.634  5539  5539 W unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["WIFI_P2P:PEERS_UPDATED",\{"devices":[\{"status":3,"primaryDeviceType":"10-0050F204-5","isGroupOwner":false,"secondaryDeviceType":null,"deviceAddress":"f2:7c:3f:99:35:60","deviceName":"realme 8s 5G"\}]\}])\
03-16 01:51:59.648  5539 11072 I ReactNativeJS: 'OnPeersUpdated', [ \{ status: 3,\
03-16 01:51:59.648  5539 11072 I ReactNativeJS:     primaryDeviceType: '10-0050F204-5',\
03-16 01:51:59.648  5539 11072 I ReactNativeJS:     isGroupOwner: false,\
03-16 01:51:59.648  5539 11072 I ReactNativeJS:     secondaryDeviceType: null,\
03-16 01:51:59.648  5539 11072 I ReactNativeJS:     deviceAddress: 'f2:7c:3f:99:35:60',\
03-16 01:51:59.648  5539 11072 I ReactNativeJS:     deviceName: 'realme 8s 5G' \} ]\
03-16 01:51:59.689   949  1055 E Light   : AmbienteLight: 116.000000\
03-16 01:51:59.694  5539 11072 I ReactNativeJS: 'Devicesssss', [ \{ status: 3,\
03-16 01:51:59.694  5539 11072 I ReactNativeJS:     primaryDeviceType: '10-0050F204-5',\
03-16 01:51:59.694  5539 11072 I ReactNativeJS:     isGroupOwner: false,\
03-16 01:51:59.694  5539 11072 I ReactNativeJS:     secondaryDeviceType: null,\
03-16 01:51:59.694  5539 11072 I ReactNativeJS:     deviceAddress: 'f2:7c:3f:99:35:60',\
03-16 01:51:59.694  5539 11072 I ReactNativeJS:     deviceName: 'realme 8s 5G' \} ]\
```

Regarding Android 13 - have you seen this thread/comment https://github.com/kirillzyusko/react-native-wifi-p2p/issues/59#issuecomment-1519228702?

Suyashbajpai27, 2024-03-18:
I know but it is in debug mode that is why i am able to send you the logcat and i found this issue is from version,like when i am trying to run the app in android version 11 or lower then it is working but it is not working in upgraded version(more than Android 11).

kirillzyusko, 2024-03-18:
> I know but it is in debug mode that is why i am able to send you the logcat

`logcat` should work on both debug and release builds.

Do I correctly understand, that you opened an issue that it doesn't work in release mode and when I asked you to share adb-logs, then you shared them for debug build?..