kirillzyusko/react-native-wifi-p2p #96: Library is not working in Android Version 12 or more upgrade version

Suyashbajpai27, 2024-03-18:
This library is not working in Android Version 12 or more upgrade version, when i am calling this method startDiscoveringPeers() getting the error in console {"code": 0, "message": "Operation failed due to an internal error."}.Please try to resolve this.

kirillzyusko, 2024-03-18:
@Suyashbajpai27 can you attach adb logs? Typically before "Operation failed due to an internal error." Android will log an actual reason why it failed.

Suyashbajpai27, 2024-03-18:
I am attching my ADB logcat , can you please help me to solve this error,
[Adb Logcat New.txt](https://github.com/kirillzyusko/react-native-wifi-p2p/files/14640812/Adb.Logcat.New.txt)

kirillzyusko, 2024-03-19:
@Suyashbajpai27 looks like you didn't ask for permissions?

```
<p class="p1">03-18 23:43:09.256<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1846 W ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcastMultiplePermissions:1232 android.content.Context.sendBroadcastWithMultiplePermissions:2363 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendBroadcastMultiplePermissions:4510 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendP2pConnectionChangedBroadcast:4552 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.-$$Nest$msendP2pConnectionChangedBroadcast:0<span class="Apple-converted-space"> </span></p>
<p class="p1">03-18 23:43:09.257<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1846 W ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:1297 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendBroadcastMultiplePermissions:4524 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendP2pConnectionChangedBroadcast:4552 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.-$$Nest$msendP2pConnectionChangedBroadcast:0 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine$P2pEnabledState.enter:2486<span class="Apple-converted-space"> </span></p>
<p class="p1">03-18 23:43:09.257<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1659 D WifiDisplayController: Received WIFI_P2P_CONNECTION_CHANGED_ACTION: networkInfo=[type: WIFI_P2P[], state: UNKNOWN/IDLE, reason: (unspecified), extra: (none), failover: false, available: false, roaming: false]</p>
<p class="p1">03-18 23:43:09.258<span class="Apple-converted-space">  </span>2168<span class="Apple-converted-space">  </span>2368 I Tethering: WifiP2pAction: P2pInfo: groupFormed: false isGroupOwner: false groupOwnerAddress: null Group: null</p>
<p class="p1">03-18 23:43:09.259<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1678 W BroadcastQueue: Appop Denial: receiving Intent { act=android.net.wifi.p2p.CONNECTION_STATE_CHANGE flg=0x4000010 (has extras) } to ProcessRecord{7249c11 1626:system/1000} (pid=1626, uid=1000) excludes appop android:fine_location due to sender android (uid 1000)</p>
<p class="p1">03-18 23:43:09.259<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1678 W BroadcastQueue: Appop Denial: receiving Intent { act=android.net.wifi.p2p.CONNECTION_STATE_CHANGE flg=0x4000010 (has extras) } to ProcessRecord{7249c11 1626:system/1000} (pid=1626, uid=1000) excludes appop android:fine_location due to sender android (uid 1000)</p>
<p class="p1">03-18 23:43:09.259<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1678 W BroadcastQueue: Permission Denial: receiving Intent { act=android.net.wifi.p2p.CONNECTION_STATE_CHANGE flg=0x4000010 (has extras) } to ProcessRecord{b1b19ac 2168:com.android.networkstack.process/1073} (pid=2168, uid=1073) requires android.permission.NEARBY_WIFI_DEVICES due to sender android (uid 1000)</p>
<p class="p1">03-18 23:43:09.259<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1678 W BroadcastQueue: Permission Denial: receiving Intent { act=android.net.wifi.p2p.CONNECTION_STATE_CHANGE flg=0x4000010 (has extras) } to ProcessRecord{8b7466d 14612:com.sampleproject/u0a253} (pid=14612, uid=10253) requires android.permission.NEARBY_WIFI_DEVICES due to sender android (uid 1000)</p>
<p class="p1">03-18 23:43:09.263 14612 14739 E gralloc4: Empty SMPTE 2094-40 data</p>
<p class="p1">03-18 23:43:09.269<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1846 W ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcastMultiplePermissions:1232 android.content.Context.sendBroadcastWithMultiplePermissions:2363 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendBroadcastMultiplePermissions:4510 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendThisDeviceChangedBroadcast:4533 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.updateThisDevice:5625<span class="Apple-converted-space"> </span></p>
<p class="p1">03-18 23:43:09.269<span class="Apple-converted-space">  </span>1626<span class="Apple-converted-space">  </span>1846 W ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:1297 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendBroadcastMultiplePermissions:4524 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.sendThisDeviceChangedBroadcast:4533 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.updateThisDevice:5625 com.android.server.wifi.p2p.WifiP2pServiceImpl$P2pStateMachine.initializeP2pSettings:5608<span class="Apple-converted-space"> </span></p>
```

kirillzyusko, 2024-03-19:
Can be a duplicate of https://github.com/kirillzyusko/react-native-wifi-p2p/issues/59

kirillzyusko, 2024-03-19:
@Suyashbajpai27 Starting from Android 6 it's not sufficient to simply declare permissions in `AndroidManfest.xml` - you need to ask permission in runtime. I wrote about it here: https://github.com/kirillzyusko/react-native-wifi-p2p?tab=readme-ov-file#caveats

The best way would be to use `react-native-permissions` library and ask additional permissions: https://github.com/zoontek/react-native-permissions

Samples of the code can be found in the link https://github.com/kirillzyusko/react-native-wifi-p2p/issues/59

Suyashbajpai27, 2024-03-20:
Thank you so much for this but now i am getting issue while using cancelConnect() , then i am getting error "Something gone wrong. Details:  {"code": 2, "message": "Operation failed because the framework is busy and unable to service the request."}", Can you please help me to get out of this?

kirillzyusko, 2024-03-21:
> then i am getting error "Something gone wrong. Details: {"code": 2, "message": "Operation failed because the framework is busy and unable to service the request."}", Can you please help me to get out of this?

Please, open a new issue and attach logcat logs 🙏

Suyashbajpai27, 2024-03-21:
Okay sure

closed 2024-06-21 by kirillzyusko