kirillzyusko/react-native-wifi-p2p #98: Library not working on latest android versions

MadhavNimavat, 2024-06-20:
I have created a demo based on the code given and when I'm trying to run on Android 13 it not working as expected getAvailablePeers() function always returns an empty array. it not finding devices where I have two emulators .

kirillzyusko, 2024-06-21:
Hey @MadhavNimavat 

Did you read [caveats](https://github.com/kirillzyusko/react-native-wifi-p2p?tab=readme-ov-file#caveats) section?

c-info, 2024-06-22:
yes @kirillzyusko I have updated all required code also I'm trying to run the demo code provided in the readme

kirillzyusko, 2024-06-24:
@c-info the example project has out-of-date code and I wrote it only when Android 9 was available. You'll need to modify the code yourself and ask all necessary permissions 🤷‍♂️

c-info, 2024-06-24:
yes i have added all necessary permissions. Also let me know will it work by using two emulators ?

kirillzyusko, 2024-06-24:
> Also, let me know if it will work by using two emulators.

Nope, never - only real devices and I wrote about that in the beginning of the README.

Please, provide adb logs, so that I can assist further