kirillzyusko/react-native-wifi-p2p #99: Not working on Andriod 14

ganesh42shrma, 2024-06-21:
Hey I am new to react-native , but I followed your documentation and did the necessary imports and also ensured I specified all the permissions in androidmanifest.xml as well used the react-native-permissions module in run time. When I run the Application i get the following errors ( please refer the image and the logcat.txt ) Please help and suggest changes. Thank you
[apptsx.txt](https://github.com/user-attachments/files/15928305/apptsx.txt)

[logcat of wifidirect example.txt](https://github.com/user-attachments/files/15928307/logcat.of.wifidirect.example.txt)

[image]

[image]. 
I have also provided my App.tsx code just in case.

kirillzyusko, 2024-06-21:
Looks like a module hasn't been linked?

Have you tried to reassemble the app?

ganesh42shrma, 2024-06-24:
The auto-linking feature is not working so I will try to link it manually. Could you clarify:
Recommended React Native version?
Does "new architecture" mean no arm64 support?
Should I use an arm device instead?
Are there plans for arm64 support?

kirillzyusko, 2024-06-24:
@ganesh42shrma it supports all Android architectures. Right now RN has two architectures (paper - old, fabric - new), and this package supports only old (paper) architecture.

> Recommended React Native version?

Theoretically it should be compatible with all recent RN versions 🤷‍♂️

ganesh42shrma, 2024-06-24:
Hey , The application did launch but  it returns an empty array of devices. The permissions are also now granted ig ( that wasnt the case earlier ) . I have attached the adb logcat for the same, could you help me out please .

[image]
[image]
[image]
[image].  
[abd logcat.txt](https://github.com/user-attachments/files/15952120/abd.logcat.txt)
[abd logcat.txt](https://github.com/user-attachments/files/15952120/abd.logcat.txt)

kirillzyusko, 2024-06-24:
In your logs I see:

```bash
2024-06-24 13:53:58.501 26086-26152 ReactNativeJS           com.wifidirectapp                    I  Permission denied: P2P mode may not work correctly.
```

Maybe because of that it returns an empty array?

kirillzyusko, 2024-06-24:
Also it looks like logs are grouped for a specific process `com.wifidirectapp`? I would suggest to grab all logs, because sometimes OS throws useful logs that clearly indicates where the problem is 👍

ganesh42shrma, 2024-06-24:
Sorry to bother you again, I went through the logs , I was unable to debug why this is happening , I have also attached all logs . Maybe you can find something ? 
[All_logs.txt](https://github.com/user-attachments/files/15953156/All_logs.txt)

ganesh42shrma, 2024-06-24:
This keeps popping up , maybe it has something to do with not getting the available devices ?
```
2024-06-24 15:43:05.788  1076-1076  wificond                wificond                             E  Failed to get interface index from scan result notification
```

ganesh42shrma, 2024-06-24:
Actually  , I can see the devices now. Tested it on different devices , somehow it did not work on one device , will troubleshoot further.Thank you for responding immediately and helping me out. I will ask for your help in the future again if I have any bugs,issues regarding the module please don't 

[image]
mind.

closed 2024-06-24 by ganesh42shrma