kirillzyusko/react-native-wifi-p2p #102: getAvailablePeers() is not showing up the available devices when testing in physical devices.

FounderBrkandcnct, 2025-01-28:
I've installed the release apk in two different mobile phones where wi-fi direct option is available. But getAvailablePeers() is not showing anything onto the screen.

you can check the code in App.tsx file .

https://github.com/FounderBrkandcnct/poster/

FounderBrkandcnct, 2025-01-29:
any update @kirillzyusko

FounderBrkandcnct, 2025-01-29:
any update @viniciuscb @qalqi @seba9999 @tyanbiao

closed 2025-01-29 by FounderBrkandcnct

kirillzyusko, 2025-01-29:
@FounderBrkandcnct did you find a solution for this problem?

FounderBrkandcnct, 2025-01-29:
yep. but i got another issue . if i'm trying to access deviceName / deviceAddress from the code to uniquely identify the device for my project , i'm not able to access it since android restricts due to its privacy policies. Do you know any other unique way to identify each device pal ?

kirillzyusko, 2025-01-31:
>  if i'm trying to access deviceName / deviceAddress from the code to uniquely identify the device for my project , i'm not able to access it since android restricts due to its privacy policies. Do you know any other unique way to identify each device pal ?

According to android docs it's still should be possible to connect with [`deviceAddress`](https://developer.android.com/develop/connectivity/wifi/wifi-direct#java) 🤷‍♂